florianfesti/boxes #1: Artifact in surroundingWall if y=2*radius

MAKOMO, 2016-03-24:
First of all thanks for this wonderful code. This is extremely helpful! 

Just a small thing: I discovered a tiny artifact in a surroundingWall rendering in case one generates a box with two 180° rounded sides, i.e. y=2*radius. Here the straight part between the two 90° round corners gets zero and I expected to end up with a clean 180° flex wall, just twice as long as each single 90° wall, with nothing in between. However, a small distance seems to be generated in between those two round corners.

That screenshot might explain it better:

[image]

My quick fix is to test if l-2*r get's zero in boxes.py:surroundingWall by replacing

```
        for i, l in zip(range(4), (y, x, y, 0)):
            self.flexEdge(c4, h+topwidth+bottomwidth)
            self.cc(callback, i+1, y=bottomwidth)
            if i < 3:
                bottom(l-2*r)
```
with
```
        for i, l in zip(range(4), (y, x, y, 0)):
            self.flexEdge(c4, h+topwidth+bottomwidth)
            self.cc(callback, i+1, y=bottomwidth)
            if i < 3 and l-2*r:
                bottom(l-2*r)
```

But there might be another way to handle this better.

Model to reproduce:

```
from boxes import Boxes, FingerJointSettings
import math

class Bug(Boxes):
    def __init__(self):
        Boxes.__init__(self)
        self.buildArgParser("x", "y", "h")
        self.argparser.add_argument(
            "--radius",  action="store", type=float, default=41,
            help="Radius of the latch in mm")

    def render(self):
        x, y, h, r = self.x, self.y, self.h, self.radius
        t = self.thickness
        b = self.burn        
        c4 = (r+self.burn)*math.pi*0.5     
        width = 4*c4 + 2*(x-2*r) + 3*t + 2*b
        height = h + 5*t + 10*b        
        self.open(width, height)        
        self.surroundingWall(x,y,r,h)
        self.close()        
        
if __name__ == '__main__':
    b = Bug()
    b.parseArgs([
        "--x", "150", 
        "--y", "78", 
        "--h", "40",
        "--radius", "39",
        "--output", "bug.svg"])
    b.render()

```

Two more small remarks:
- wouldn't it make sense to separate the parameter burn from the ctx line_width as some machines ignore non-hairlines, larger than 0.01. For now I just use a self.ctx.set_line_width(0.01) after open()
- in my first ever cut on a laser I found the default 10% stretch for the flex part of a surroundingWall too short. Wouldn't it make sense to also turn this stretch factor into a parameter? 

Thanks for sharing your code!

florianfesti, 2016-03-29:
Ok, fixed in current head. I now just merge the two FlexEdges into one. Drawback is that this also kills the callbacks for the y sides, but as there is no space to draw anything anyway may be no one cares...

There are a couple of details like the ctx line_width that probably deserve a config file. But there has just not been enough stuff for me to bother with that.

Actually there is a stretch setting in the FlexSettings already. It is just not used yet...

closed 2016-03-29 by florianfesti

MAKOMO, 2016-03-29:
Thanks for your quick reaction. Excellent solution, no one will miss that callback. Keep going!

MAKOMO, 2016-03-30:
> Actually there is a stretch setting in the FlexSettings already. It is just not used yet...
So what about replacing the line

`c4 = 0.9 * c4 # stretch flex 10%`

in `boxes/__init__.py` by

by

`c4 = (1.0-(self.stretch/100.0)) * c4`

in `surroundingWall` to activate that setting?

It would also be cool if one could specify alternative left and right edges in surroundingWall (in addition to the bottom and top ones) to replace the default doveTailJoint pair.

florianfesti, 2016-03-30:
Done. Although I just use c4 / stretch and keep stretch in the range of 1.0 to 1.1

MAKOMO, 2016-03-30:
You are amazing! Now I can define my "box" without any monkey patching. Great tool. Excellent API!
Thanks a lot for let me use this for free.

MAKOMO, 2016-04-09:
Florian, sorry for abusing this channel in the lack of an alternative way (opening another issue would not be appropriate either) to contact you. I wonder what would be the correct way to use alternative settings temporary for some edges. 

I see from your example in the template that for regular edges I could/should generate an instance of one of the given edge classes, applying the corresponding settings and potentially assigning it a character using addPart.

However, how could I influence the settings used within the fingerHolesAt "macro"?

Again thanks for your work on this extremely useful lib! I am trying currently to make my "custom" generators robust over your further enhancements, being aware that the lib is quite under intensive development and APIs are not fully stable.

Thanks
Marko

florianfesti, 2016-04-09:
Opening new issues is actually the right and my preferred way of communication for issue like this one.

Generally creating new Edge instances with their own Settings instance is the right way to create Edges with different settings. This currently does not work properly with FingerHolesAt as it is not really an Edge class. So I will need to find a way to work around this. I'll probably change the signature of the FingerHoles.__call__ to match fingerHolesAt(). The you can just instantiate your own version of FingerHolesAt.

For now you can just create a FingerHoles instance sharing the new Settings instance and wrap it with your own At() part.

MAKOMO, 2016-04-09:
> Opening new issues is actually the right and my preferred way of communication for issue like this one.
Ok I will in the future. Just I did not feel this is an issue nor a feature request

Regarding the fingerHolesAt() I will follow your advice and for now add my own At() part to a corresponding FingerHoles instance.

For changing edge settings temporarily I thought a with_settings(s) like macro would be nice, but that just breaks your concept of having edges with different settings as separate instances.

florianfesti, 2016-04-09:
OK, sorry to probably breaking your code again.
I now use the FingerHoles class every where which then holds the reference to the Settings. On the bright side your At() part can probably go away again.

I hope this doesn't break things too badly...

MAKOMO, 2016-04-09:
Excellent! Don't worry on braking things on my side. My "things" are rather simple and I love to see how this develops into an extremely flexible tool. Keep going!