florianfesti/boxes #3: DoveTailJointCounterPart in CompoundEdge

MAKOMO, 2016-04-22:
```
        self.open(50, 50)
        x = y = 10
        r = 0
        h = 46
        sep = 20 # 12
        self.surroundingWall(x,y,r,h,
            left=edges.CompoundEdge(self,["D","e"], [h-sep,sep]),
            right=edges.CompoundEdge(self,["e","d"], [sep,h-sep]))
        self.close()
```

renders

[image]

The left edge is not what I expected. The right edge looks fine.

MAKOMO, 2016-04-22:
Oh, and if I add `bottom='F'`I expected the total hight to increase by `thickness`. Isn't the height argument interpreted as the inner height? At least I was under the impression that some commits before that was the case. I might be wrong.

ffesti, 2016-04-26:
Well, the surroundingWall was not really ready to have fully parametrized edges. Also nice catch on the "F" width mess up. Both should be fixed now.

MAKOMO, 2016-04-26:
I confirm that the "F" width is back to normal with our latest commit. The compound left edge improved somewhat, but does still not render as I would expect it.

[image]

Why is there this 90 degree turn to the inside after the "D" edge? I would have expected just a straight line as on the right side.

MAKOMO, 2016-04-26:
Ah, now I found where that 90 degree turn to the inside happens. It is in CompoundEdge/__call__ in the elif part. However, I get what I expect if I comment the "def startwidth" in DoveTailJointCounterPart.

florianfesti, 2016-04-26:
Yup, the DoveTailJointCounterPart should be width zero so that the DoveTailJoint doesn't need any length of the surrounding wall. Fixed now.

MAKOMO, 2016-04-26:
Yes, I think this issue is now closed, however, I now have another "CompoundEdge issue" in another place. I will investigate and potential have to open another issue...

closed 2016-04-26 by MAKOMO