florianfesti/boxes #4: CompoundEdge with FingerJointEdge/FingerJointEdgeCounterPart Part

MAKOMO, 2016-04-26:
This might be just a stupid user misunderstanding of the specs. However, I expected that one of the two variants (FingerJointEdge or FingerJointEdgeCounterPart) would go to the outer side of the wall and the other towards the inner side. Here is my example code:

```
        self.open(55, 60)
        x = y = 10
        r = 0
        h = 46
        sep = 20        
        
        s = edges.FingerJointSettings(self.thickness, space=2, finger=2, surroundingspaces=0, relative=True)
        #p = edges.FingerJointEdgeCounterPart(self, s)
        p = edges.FingerJointEdge(self, s)
        p.char = "p"
        
        self.addPart(p)
        self.surroundingWall(x,y,r,h,
            left=edges.CompoundEdge(self,["e","p"], [2*h/3.0,h/3.0]),
            right=edges.CompoundEdge(self,["p","e"], [h/3.0,2*h/3.0]),
        )
        self.close()
```

The one variant looks like this

[image]

and the other like this.

[image]

Both turn to the outside of the straight wall. Is this intended?

ffesti, 2016-04-26:
Yes, this is intended. The FingerJointEdge is intended to create a 90° joint. Have a look at e.g. the box.py generator. To have a straight overlap at the corner you can use the "E" type Edge which is properly set out.

MAKOMO, 2016-04-27:
Oh I see. So this is an issue on my side. I'll take it. I thought the x-dimension should be adjusted, but it does not as x is intended to indicate the inner dimension of the box and that does not change here. I now have to check why my code still delivered what I was expecting some commits before and now it (correctly) does not anymore. Sorry for the noise.

closed 2016-04-27 by MAKOMO