florianfesti/boxes #5: fingerHolesAt seem not to respect the kerf

MAKOMO, 2016-06-10:
Dear Florian,

I finally received a laser and could start testing the models I constructed before using your fine lib. Everything works out as expected. However, one model uses fingerHolesAt in a callback to "install" a separating wall in a box. While the walls them self well-fit (a little tight thus) with the burn parameter I set, the holes generated using the fingerHolesAt macro are too small to insert the corresponding fingers.

[image]

As seen in the screenshot I tried to measure the holes and the fingers and came to the following result:

a) burn=0.06, thickness=3
- holes generated using fingerHolesAt: 2.88mm x 2.88mm
- pin distance (generated using edges["F"]): 2.94mm
- pin thickness(generated using edges["F"]): 3.06mm

b) burn=0.16 (way to large to make a fit!), thickness=3
- holes generated using fingerHolesAt: 2.68mm x 2.68mm
- pin distance (generated using edges["F"]): 2.68mm
- pin thickness(generated using edges["F"]): 3.31mm

So at burn=0.16 the distance between the fingers and the size of the hole agree (but of course this would not fit in any case as that burn value is to high), but not at burn=0.06 the size of the whole and the finger distance do not agree. The finger distance works for those fingers with the kerf of my laser as expected, but not the tighter fingerHolesAt generated hole.

Any idea?

ffesti, 2016-06-10:
That's pretty weird. Actually the finger holes are correct: They should be 3mmx3mm nominal minus two times the burn. So 2.88mm + 2* 0.06mm = 3mm is just fine.

The other measurements look strange though. The width should be 3mm+2*burn = 3.12 and the distance should be 2.88 (same reason as finger hole size). The depth should be exactly 3mm - but this is a bit tricky to measure.

The numbers for the 0.16 burn look good. I guess there is an issue with how the files are processed. Boxes.py uses 2*burn as line width. So you can see what gets cut away. It is possible that your laser software or SVG importer uses the border of the line instead of the center. You can try changing
` ctx.set_line_width(2*self.burn)`
in Boxes.__init__.py

I already read about some software stack having problems with line width but the one I used so far was just fine with it.

MAKOMO, 2016-06-10:
Hm. I will further investigate. Actually, my script is doing a 'ctx.set_line_width(0.01)' to ensure that the line is a hair line. Regarding the laser software, we purchased a Zing24, but not using the Zing software/driver, but we use the VisiCut software for cutting. Maybe everything is fine with the model, just the the burn we used was a bit too optimistic. It works for the regular finger (but the fit is quite tight) and for those holes the very same finger just don't fit. Maybe the material, Kraftplex ST 3mm, is not exact enough!? Regarding those second measurements, I might have just messed up on measuring with that wired measuring tool in Inkscape that I have a hard time to understand. I will come back on this one. 

Which software are you using for cutting? Could the problem be with VisiCut!? Why does it work for the regular fingers, but not those wholes? Something seems just to be different. I'll find out and report.

florianfesti, 2016-06-11:
Ok, one more note: Make sure to measure the thickness of the material used. Especially wood based materials often do deviate from their nominal thickness. While that might not be that critical for finger joints on the edges it will create problem for the finger joint holes.

MAKOMO, 2016-06-11:
Thanks Florian for that extra hint. I am just new to this laser cutting thing and did not expect that. The material is indeed closer to 3.15mm or even 3.2mm. That's it. I will change that in my model and retry.

The good thing is that I learned how to proper measure distances with Inkscape. The real measurements of the two models are

a) burn=0.06, thickness=3

holes generated using fingerHolesAt: 2.88mm x 2.88mm
pin distance (generated using edges["F"]): 2.88mm
pin thickness(generated using edges["F"]): 3.06mm

b) burn=0.16 (way to large to make a fit!), thickness=3

holes generated using fingerHolesAt: 2.68mm x 2.68mm
pin distance (generated using edges["F"]): 2.68mm
pin thickness(generated using edges["F"]): 3.32mm

So the boxes lib does everything perfectly right. I'll close this stupid one.

closed 2016-06-11 by MAKOMO