florianfesti/boxes #11: FlexEdge out of center

MAKOMO, 2016-07-07:
Sorry, me again.

It seems that FlexEdges are not fully entered within their length. A little hard to see in the screenshot below as one does not see exactly the start and end of the two FlexEdges. However, the holes are here fully centred within the intermediate edge, but the distance to the left flex is larger than that to the right flex.

[image]

I took a look at edges.py:FlexEdge:__call__ and saw that you already compute a leftover and use it for centering. Took me a moment to understand that you also need to compensate for the fact that the first line/cut is starting on the calculated pos, but the left line is positioned still one dist from the calculated end of the flexedge segment (this is easier to show than to express in words;).

Anyhow, the fix is to replace

`pos = i*dist + leftover/2.0`

by

`pos = i*dist + leftover/2.0 + dist/2.0`

florianfesti, 2016-07-08:
I understand the problem and it clearly needs to be fixed. I am just unsure how. There are three options which both have their issues:
1) Remove the first line - this will lead to a double width gap when flex meets flex
2) Add the + dist / 2 - this may lead to half width links if flex meets an inset edge
3) Add line at the end - may lead to even worse links or very close distance to next flex

May be I should just add a settings param as with surroundingspaces for FingerJoints.

myTonino, 2016-07-08:
Hm. 1) sounds safest as one would never end up with a half width link for that flex. Does a flex will meet a flex at all? Could those cases be avoided?

Would a settings parameter help? There would potentially be still a rest that would need to be handled?

florianfesti, 2016-07-09:
In general having two consecutive flex edges should be avoided, but this can happen if a edge in between gets to length zero without special code in place to merge then (just improved that in the flexbox* generators)

So I opted for 1) as it works in all edge cases. May be I'll add a param later to get one more cut on each side for added bending capability.

closed 2016-07-09 by florianfesti

myTonino, 2016-07-10:
Thans Florian, my flex boxes look much better now;)