florianfesti/boxes #28: Inkscape integration

florianfesti, 2017-02-18:
Make box generator available as Inkscape plugins.

This requires multiple pieces:

- Generating .inx XML files from argparse objects
- Generate menu structure (using the UIGroups)
- Make sure the way cli params are handled is compatible
- Copy generated svg into document passed from Inkscape

florianfesti, 2017-02-25:
Works but needs some more love. Especially some docs...

closed 2017-02-25 by florianfesti

misiekkk, 2017-05-04:
Inkscape integration is a very nice idea ;-)
If You need somebody to test integration with Inkscape at Windows and Linux - I can help You.
I try to compile .inx files of boxes.py by setup.py, but:
- on Windows - setup.py don't work, boxes2inkscape work, but I have problem with cairo ( python 3.6 ) 
- on Linux ( ubuntu 16.10 ) - problem with setuptools ( installed but don't work ) ( python 2.7 )

Do You have compiled .inx files ?
Can You share it ?

florianfesti, 2017-05-04:
Even without setup.py working you should be able to just execute
scripts/boxes2inkscape ~/.config/inkscape/extensions/
I just created a symlink in ~/.config/inkscape/extensions/ to ...boxes/scripts/boxes
This allows me to run everything fro my checkout. But having the setup.py work would actually be nicer...

Can you please open tickets with more detailed descriptions of how things break?

misiekkk, 2017-05-12:
Today I finish integration of boxes.py with inkscape 0.92.1 ( Python 2.7 32 bit ) 64bit on Windows 8.1 64 bit.

What I do to integrate this extension:

1. Update and prepare Python for work:
- add C;\Program Files\Inkscape\Python and C;\Program Files\Inkscape\Python\scripts to PATH Environment Variable and RESTART SYSTEM

- Run cmd as Admnistrator
         - upgrade setuptools ( pip install --upgrade setuptools )
         - upgrade svgutils ( pip install --upgrade svgutils )

2. Modify extension files:
- script/boxes2inkscape: line 16
  added: import os

- script/boxes2inkscape: line 114
  from:     <command reldir="extensions" >boxes</command>
  to:         <command reldir="extensions" interpreter="python">boxes</command>
  ( without this modification boxes don't run under inkscape )

- boxes/__init__.py: line 526
  from: svgutil.svgMerge(self.output, self.inkscapefile, sys.stdout.buffer)
  to:      svgutil.svgMerge(self.output, self.inkscapefile, sys.stdout)
  ( without this modification I get error about incorrect parameter )

3. Generate extension files
- Run cmd as Administrator
     - python setup.py install
     - python boxes2inkscape ..\inkex

4. Copy .inx files from ..\inkex to extension directory of inkscape ( local or global )

After this extension work very well.
Thank You !

Best Regards
Mariusz

florianfesti, 2017-11-13:
For the record: The fixes above made it upstream with the exception of the 
<command reldir="extensions" >boxes</command> to
<command reldir="extensions" interpreter="python">boxes</command>
change.