florianfesti/boxes #29: Make outer edge continuous

florianfesti, 2017-02-26:
even if other parts are drawn intermediately.

It's still unclear if thit can be done with cairo. Further examination is needed. May be draw on separate groups or canvases during callbacks.

dshookowsky, 2017-07-29:
[image]
[image]

I think this is affecting me when I try to use the output with laserweb.  It's skipping part of the path and almost looks like it's joining different shapes

florianfesti, 2017-12-10:
Looks like this might be possible to fix using the RecordingSurface. Unfortunately is is only available in pretty recent cairo versions and is still buggy. While I could make it work with the PDFSurface it failed miserably with the SVGSurface.
I was playing with the RecordingSurface to get a sane way of determining the size of the drawing and to enable the built-in PDF back end.

Mumeii, 2018-03-29:
On my side i only use boxes.py thanks the web interface, save the produced svg file and load it with inkscape. Doing so I've indeed noticed that produced boxes outer frames are not continuous due to strange inkscape behaviour while performing union, division or such path operations on produced boxe's paths

Having a deeper look at it, I've noticed that Inkscape perform path formulas optimizations, like for example `v 0` useless shortcut or `m 0 0` relative useless moves, that split the path apart

I guess the root cause is unoptimized path formulas produced by Cairo such as ::

`... L 5.5 0 M 5.5 0 ...` (pencil useless move)

fixing them with a regular expression by removing the useless Move operation lead to further possible optimisations on things such as ::

`... C 1 2 3 4 L 3 4 ...` (blank line drawing)

So right now i'm using the following quick fix as i'm definitely not familiar with either Python or Cairo. I apply twice the following regular expression replacement on the produced file ::

`(\-?\d+(\.\d+)?) (\-?\d+(\.\d+)?) (M|C|L) \1 \3` replaced with `$1 $3`

Hope it can help a bit and you could patch the code and the website in a near future ;)

florianfesti, 2019-07-24:
Path is now optimized by removing unnecessary moves (#63).
Still need to check whether the path gets interrupted by callbacks. May be the pieces can be stitched together by looking for paths with matching end and start point.

florianfesti, 2020-05-19:
OK, the new back end (see #192) this should be solved. It indeed could not be done with Cairo.

Closing.

closed 2020-05-19 by florianfesti