florianfesti/boxes #32: Use stretch setting in flexboxes

florianfesti, 2017-02-26:
While we have a parameter in the flex settings it is often not used. May be the flex edge should use is directly. But this would require an opposing edge that does the same thing.

MAKOMO, 2019-12-09:
Maybe a start could be to add the line

`c4 = c4 / self.edges["X"].settings.stretch`

also to the definitions of surroundingWall (the same way as in that definition in __init__.py) in the flexbox generators!?

florianfesti, 2019-12-09:
Yes, the coding part is pretty trivial. This is more a question whether the generators still work after the change and whether the defaults are sane.

matkoniecz, 2021-03-16:
If it is not used and flexboxes work, maybe set default to 1 (what should have not effect if my guess that it is multiplier is right)

That would make it far less confusing, I am trying to debug why my first box is a failure :)