florianfesti/boxes #36: Add settings for Bedbolts

florianfesti, 2017-02-26:
While the library supports "bed bolts" (slots for nuts and screws holding walls together at joints) the generators don't.

miguelmorin, 2018-07-24:
Hi! Well done with the project. Your presentation has a "Box2.py with bedbolts". I could not find such a file in the repo. So can I produce files for a box with bedbolts, or do I have to wait for this issue to close?

florianfesti, 2018-07-24:
Box2 got renamed to UniversalBox - which already has a settings page like a cockpit of a 747. So that's probably not the right place to start with this feature.

If you want to toy around with bed bolts you can remove the following line from boxes/generators/closedbox.py:
`d2 = d3 = None`

The issue with this ticket is that I only use ply wood for my boxes which works just fine without bed bolts. And I really don't know what settings one would want/need to make them useful. If you are able to provide some input adding the code may be not that difficult.

florianfesti, 2018-07-25:
You should at least see some bed bolt slots. I "fixed" the closed box to have them at all the right places - not just somewhere random.

miguelmorin, 2018-07-25:
Mmm, I have these lines in `boxes/generators/closedbox.py`:
```
        d2 = edges.Bolts(2)
        d3 = edges.Bolts(5)

        #d2 = d3 = None
```
And I see no bed bolts, as in this image where Inkscape emphasizes the anchor points of the path for one of the sides:
[image]

Do you get bed bolts when you comment out line 45?

florianfesti, 2018-07-25:
Yes.

miguelmorin, 2018-07-27:
I think I have the wrong setup. If I rename `closedbox.py` to `closedbox2.py` or add a false assertion in the class:
```
class ClosedBox(Boxes):
    """Fully closed box"""

    assert False, "Does the script hit this line?"
```

then the call

```
scripts/boxes closedbox --thickness=5 --output=test.svg --format=svg --x=254 --y=300 --h=100
```

still produces a file, instead of an error. @florianfesti can you tell me your setup?

florianfesti, 2018-07-27:
Hmm, looks like it imports the files from somewhere else. May be you installation in the the main Python instance. You can do import boxes.generators.closedbox in the Python shell and then print(boxes.generators.closedbox.__file__)

florianfesti, 2018-07-27:
No. This is probably a coincident that I have not run into this problem.
I added
`try:
    import boxes
except ImportError:
    sys.path.append(os.path.dirname(os.path.realpath(__file__)) + '/..')
    import boxes
`
to make sure the sources are found even when not being installed. But I always call  `scripts/boxesserver` from the repository main dir. As the current dir is first in the path for Python that always leads to be getting the code from the repo. You can either do the same thing or adjust your Python path.

miguelmorin, 2018-07-27:
I'll make another suggestion: to verify that the installed module `boxes` has a `__file__` not `None`. I tried multiple times and uninstalled the module as in [this thread](https://stackoverflow.com/questions/1550226/python-setup-py-uninstall). Uninstalling removed the files but left a directory structure `/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/boxes-0.1-py3.7.egg/` with no files, and Python succeeded in importing an empty installed module `boxes`. I uninstalled it cleanly and now both `scripts/boxes` and `scripts/boxesserver` work and load the source files, which I can modify with the first suggestion of commenting out `d2 = d3 = None`.

miguelmorin, 2018-07-27:
I was able to adapt the `closedbox` generator with counter bed bolts, as I mentioned in #73. Thanks a lot for your help with this debugging!