florianfesti/boxes #40: Support for non-box paths

objectiveSee, 2017-05-15:
Hey! This library looks interesting. I am trying to find software that can generate the walls for a box with a complicated shape. The "box" is actually a heart shaped path (see image, which includes my own box notches). Do you think your software would work for this? I have seen other tools not work because of the type of corner marked with a green arrow. I wrote my own "notcher" library in JS but it is not as robust as I had hoped. 

[image]

florianfesti, 2017-05-15:
The problem with this corner is that the edge is shorter than they would be nominally. The easiest way out is using only finger edges on the top and bottom parts and have the "F" style edge on the top and bottom of the rectangular side walls.

florianfesti, 2017-05-15:
Hmm, this still doesn't really solve the issue.
.edgeCorner() might just do the "right" thing to make this work - although not in a nice way: elongating the edges at the corner inward one thickness.

florianfesti, 2018-09-10:
The right solution would probably be something like Boxes.polyline() but with edges. Either one Edge type all around or giving type and length tuples. The same code would generate both the outline and the side walls.

florianfesti, 2019-03-24:
Ok, 7aec26e8939057546aa52b30c525078e4609dc03 adds polygonWall() and polygonWalls() which do exactly this - including support for rounded corners with flex.
As an example ce8576d39e201cc010181f177bcd372afa28588b adds a new heart box generator.
Will updated the web instance after a bit more testing.

florianfesti, 2019-06-28:
HeartBox makes use of this new feature - although using flex for a smoother  outline.

closed 2019-06-28 by florianfesti