florianfesti/boxes #43: Angle Box menu doesn't pop up in knscape

vongomben, 2017-09-20:
Hi. 
* I generated the inx files locally, finding them in ```../boxes/inkex/```
* I copied ```boxes.py.AngledBox.inx``` both in ```/usr/share/inkscape/extensions/``` and
```~/.config/inkscape/extensions/```
* I opened Inkscape 9.1 and seen the exptension
* when I click on it nothing happens.

What am I doing wrong? 
I'm using Ubuntu 16.04 and Inkscape 0.9.1

Thanks for the program/software/extenction, it's simply game changing in creating stuff.

ffesti, 2017-09-20:
The extension uses
<dependency type="executable" location="path">boxes</dependency>
So it only shows up if the "boxes" executable (found in scripts/) is available to Inkscape in the path. You can either set a symlink from the extensions/ directory, install boxes.py into the system path or add the location to the PATH variable.

ffesti, 2017-09-20:
Eeek, The XML didn't make it through: Next try:

`<dependency type="executable" location="path">boxes</dependency>`

florianfesti, 2017-11-22:
OK, strace says that Inkscape does not actually look into the PATH. Try adding a symlink to the boxes script from the extensions sub dir for now.
I'll try fix the package soonish.