florianfesti/boxes #44: Secure finger joint on opposite side

florianfesti, 2017-10-07:
Extend the existing finger joint edge or create a new one that is form-locking. Make the finger stick out two thinknesses and have a notch on both sides. Render a comb like piece like 3 thicknesses in width that can be slid through the space between those slots to lock the part with the fingers in place.

Have a variant that can go on each side of an edge of a box securing both walls.

tzwilliams, 2021-09-24:
Perhaps a secure finger design like is used in this dice tower? [https://community.glowforge.com/t/fold-out-dice-box-serrated-tab-version/25815](url)

florianfesti, 2021-09-24:
That's not quite what I had in mind but this is also very interesting... and should be easy to add.

florianfesti, 2023-01-05:
For the record: Those serrated tabs are available for a while now as `barbs` `style` in the `FingerJointSettings`

florianfesti, 2023-01-05:
I do have code for `interlocking` style finger joints where the fingers are locked in place by a comb like piece on the outside. But I am not really happy with the result.