florianfesti/boxes #50: Inkscape Ubuntu issue

iamlocknar, 2017-11-19:
When I go to the extensions in Inkscape it shows the menu items but it does not show the actual functions (dropdown shows as empty)

florianfesti, 2017-11-19:
This means that Inkscape can not find the "boxes" binary. Check if it is in the path and add its location to the PATH variable (and make sure the new PATH is available to your Inkscape instance).

Another workaround can be creating a symlink to the boxes binary in ~/.config/inkscape/extensions/ or 
/usr/share/inkscape/extensions/

As I am only running Boxes.py from my development checkout I am interested in any steps that makes things working for normal users. So please report back if this fixes this for you. Thanks!

iamlocknar, 2017-11-21:
Sorry I am a bit new with Linux, inkscape so getting around is still a bit of a challenge. Let me describe what I did so maybe I can trace what I did wrong.

Did a git pull to get the project in my downloads folder.
Ran sudo python setup.py install
Checked inkscape and noticed the extension menus showed but the plugins did not populate
Checked Ubuntu path via echo $PATH
Saw that usr/bin was a recognized variable so manually created a bin in my usr.
Moved all of the boxes files to usr/bin/

Hate to make you troubleshoot but id LOVE to be using your plugin within inkscape. Any suggestions?

florianfesti, 2017-11-21:
Hmm, first there is a small but significant difference between usr/bin und /usr/bin
/usr/bin should exist on your system and actually the boxes binary should have been installed there.
Can you execute it by just typing boxes in a terminal?

iamlocknar, 2017-11-21:
Ok follow up! It does run in terminal. I checked the installer log and it was put in /usr/local/bin

iamlocknar, 2017-11-21:
So I'm guessing... some kind of permissions thing with inkscape accessing the local bin

iamlocknar, 2017-11-21:
SOLVED - Dropped a copy of the binary into the inkscape extensions folder. Seems to be working now!

florianfesti, 2017-11-21:
Have fun!

closed 2017-11-21 by florianfesti