florianfesti/boxes #51: Dogbones holes for milling

spiffcow, 2017-12-02:
I really like your idea, but would want to use it for milling.  Is there any reason this would be difficult?  I'm considering implementing this myself if not.

florianfesti, 2017-12-02:
Be aware that the way Boxes.py does inner corners is not ideal for milling. It is described on slides 34 to 36 in 
https://github.com/florianfesti/boxes/blob/master/documentation/presentation.odp
It might be OK for small diameter bits.

closed 2017-12-02 by florianfesti

spiffcow, 2017-12-02:
The math for a bit radius is pretty easy...  IIRC it ends up being a hole at each corner offset by (sqrt(2)-1)*r, where r is the radius of the bit.  I'll play with it and submit a PR if it works out

florianfesti, 2017-12-02:
The problem is not adding the hole but removing the overreach. If you can find an easy was of fixing it - even better.