florianfesti/boxes #52: Export to PDF or DXF

Lt-Lemming, 2017-12-07:
Most of the software I've seen for working with laser cutters doesn't support SVG or PS...

Would really appreciate if you could implement DXF or PDF as an output option as theres no good path to go from SVG to DXF that I've been able to find.

florianfesti, 2017-12-07:
Boxes.py already supports creating both DXF and PS. But you need to install the ps2edit utility (which is currently not installed on the server hosting the online generator).

As an alternative you can use Inkscape to convert from SVG to DXF.

closed 2017-12-07 by florianfesti

florianfesti, 2017-12-07:
I was typing too fast. It supports SVG and PS without anything but AI, DXF, gcode,  PLT and PDF with the ps2edit utility.
As cairo supports PDF as an output format I could add pdf as an default output format. But right now there is no way to crop the result to a sane size.