florianfesti/boxes #53: Please tag a version

greenaar, 2017-12-09:
Could you tag a version when you reach a 'stable' point?

Right now pulling from master results in alternating working and broken releases.  Tagging would make successful working milestones easier to track.

Thanks!

florianfesti, 2018-09-08:
Yes, this is how good development is supposed to work. Unfortunately that's not quite how Boxes.py is done. I do this in the spare time and add stuff whenever I feel like it or the need arises. To have a stable release I would need to have some beta first and have it tested.

In general I try to get features complete and working before I push them to the repository. But unless there is a test instance where errors can be found there is no real benefit in marking a particular commit as "good". That may change in the future but for now I just don't see this happening.

Closing. Sorry.

closed 2018-09-08 by florianfesti

greenaar, 2018-10-18:
No problem. I appreciate your response regardless.