florianfesti/boxes #54: Label h, y, and h correctly.

wizaerd, 2017-12-29:
In the online app for this, some boxes have their x, y, and h incorrectly labeled.  For example, in the FlexBox 2, the width labeled as X isn't really the width of the resulting box at all, and as such can mess people up when trying to design a box with specific dimensions.  To get a wider box, I had to increase the h setting, which is labeled as height but does not actually control the height of the box, but the height of the pattern, which corresponds to the width of the overall box.  Maybe make the fields better define what dimension of the box it is for?  Or show a diagram?

Mekol, 2018-02-08:
good idea, integratedHingeBox also needs 'x' and 'y' to be swapped

florianfesti, 2018-02-11:
Yes, it is a mess. Some of the generators were done  bit hastily.
I am thinking about replacing x, y, h by width, depth, height.
Anyway patches welcome - even for simple x, y, h swaps.

wizaerd, 2018-02-14:
A quick fix might be on the page where you enter the values, display an image of the box with labels or arrows for the x, y, and h values...  Trying to use the FlexBox2 for a recipe box has been quite a lot of trial and error to get the box in the right dimensions.

florianfesti, 2018-02-20:
Ok, fixed FlexBox2 and IntegratedHingeBox. Please test if anything broke.

Feel free to reopen if any other generators need fixing.

closed 2018-02-20 by florianfesti