florianfesti/boxes #57: IntegratedHingeBox: wried edges

Mekol, 2018-02-03:
[image]

Hey, i ve got wried edges while using http://www.festi.info/boxes.py/index.html 
My plywood cutting service doesnt like it :cry:

florianfesti, 2018-02-03:
In general Boxes.py uses very short line segments to do the burn correction aka kerf. There are know issues if the laser software uses a "merge lines" feature with a too large error margin (0.1mm often is the default). Using an margin of 0.01mm often solves this.
Does this issue only show up at this hinge? If this is the cause similar problems are expected at each inner corner.

Mekol, 2018-02-03:
@florianfesti you mean 'burn correction' ?

florianfesti, 2018-02-03:
Ok, after thinking a bit more this is not an issue with import and merging line segments.
The lines - although not looking very convincing - are actually mean to be that way.

They are used for burn correction given by the "burn" parameter which compensates for the material removed by the laser. On a mill this would be called tool compensation but the laser cutter doesn't have a tool...

The 180 arc on the left changes from kerf above the line to below.
The mess on the right is comprised of a 90 degree corner ro the right as part of the edge coming from above. Then the hinge code has to do a U-turn to outset the arc of the hinge.

So not really sure how to make this work for you.

closed 2018-02-04 by Mekol