florianfesti/boxes #60: Box idea

Mekol, 2018-03-14:
Hey, i have got a box idea, but i dont know how to script that (also i have windows and this doesnt works for me).
The main idea is to make open-box with different heights of front and back walls.
I tried to make simple sketch for this  ;)
[image]

ffesti, 2018-03-14:
Yeah, this probably deserves a generator of it's own. For now you can use a single "TrafficLight" turned upwards. Just delete the front and the FingerJointHoles for it.

florianfesti, 2018-03-15:
Try https://www.festi.info/boxes.py/UnevenHeightBox

Mekol, 2018-03-15:
Looks great, thanks for adding feature. I will test it. 😉

florianfesti, 2018-03-15:
Added param for bottom_edge

florianfesti, 2018-03-16:
Ok, tested and looks good. Two with low, high, low, high can be stacked on top of each other.

closed 2018-03-16 by florianfesti

florianfesti, 2018-03-17:
Added optional lid.

Mekol, 2018-03-17:
something is wrong using default settings(with lid)
​

Mekol, 2018-03-17:
[image]