florianfesti/boxes #61: Font errors

kakalot, 2018-03-27:
[image]

Hello, as you see it generated with font errors on any templates. Can you fix it? Thank you!

ffesti, 2018-03-27:
Yes, this is a known problem. Unfortunately I have not been able to figure out what's the problem. It looks like something on the server running the web interface broke. I already tried setting the font explicitly but nothing helped.
You can run boxes.py locally on a more modern Linux system and the fonts are rendered just fine.
If you do not have a Linux computer at hand I can generate the file for you if you give me the desired thickness and burn.

kakalot, 2018-03-29:
Hi, thank you for surporting. No problem, i will fix in svg files for myself

florianfesti, 2018-08-14:
Just for the record. The server is running CentOS Linux release 7.5.1804 (Core) 
with httpd-2.4.6-80.el7.centos.1.x86_64, mod_wsgi-3.4-12.el7_0.x86_64, python-2.7.5-69.el7_5.x86_64 and cairo-1.14.8-2.el7.x86_64. The error happens with both pycairo and cairocffi *sigh*

florianfesti, 2018-09-17:
Ok, I've been playing around with moving the instance into a more modern docker container. This fixes the issue. Still need to do a bit of engineering to set this up properly. Still closing this ticket now.

closed 2018-09-17 by florianfesti