florianfesti/boxes #62: Have finger joint smooth corners been optional

Mumeii, 2018-03-29:
As i often rework produced paths to include inner cuts in my boxes sides, the extra optimisation of having rounded corners to make perfect corner cuts is bugging me more than it helps, and specially the ones in the finger inner corners ::

[image]
Would it be possible to have this feature been optional ?

closed 2018-03-29 by florianfesti

Mumeii, 2018-03-29:
but from a pragmatic point of view, i don't understand why the laser need to perform such tiny loops while cutting the inner fingers corners. To me the paths thickness represent the laser kerf itself, thus those loops seems to uselessly remove extra material within the box face.

Is it intentional in order to ease the finger jointing process or is it due to the difficulty to distinguish cases when calling the corner function ?

Notice that, on the other hand, I'm perfectly fine with the rounded corners at finger tips as in that case it can ease matching faces together.

Anyway, i can perform a regular expression transformation step to fix it on my side, but it's a pain in the ass to have to manage it myself, lazy programmer that i am 😉

Mumeii, 2018-03-29:
just in case, here are the two regular expression to apply in order to have those rounded corners disappearing :: 

`L (\-?\d+(\.\d+)?) (\-?\d+(\.\d+)?) C \1 (\-?\d+(\.\d+)?) (\-?\d+(\.\d+)?) (\-?\d+(\.\d+)?) (\-?\d+(\.\d+)?) \9 ===> L $1 $9

L (\-?\d+(\.\d+)?) (\-?\d+(\.\d+)?) C (\-?\d+(\.\d+)?) \3 (\-?\d+(\.\d+)?) (\-?\d+(\.\d+)?) \7 (\-?\d+(\.\d+)?) ===> L $7 $3`

As you'll notice, i didn't took time to distinguish finger tip cases from finger corners cases 😞

Mumeii, 2018-03-29:
And further optimization can be performed thanks to ::

`(\-?\d+(\.\d+)?) (\-?\d+(\.\d+)?) L (\-?\d+(\.\d+)?) \3 ==> $1 $3 H $5`

`(\-?\d+(\.\d+)?) (\-?\d+(\.\d+)?) L \1 (\-?\d+(\.\d+)?) ==> $1 $3 V $5`

`H (\-?\d+(\.\d+)?) L \1 (\-?\d+(\.\d+)?) ==> H $1 V $3`

`V (\-?\d+(\.\d+)?) L (\-?\d+(\.\d+)?) \1 ==> V $1 H $3`

But the last two formulas won't work both due to a strange rounding problem in the produced box paths. Cairo trouble ?

phrxmd, 2019-11-07:
Regexps won't work anymore (changes to Cairo?)

florianfesti, 2020-03-12:
@Mumeii Sorry for the late reply!

Boxes.py does use some of these regexes itself. That may be the reason why they no longer apply. See https://github.com/florianfesti/boxes/blob/master/boxes/svgutil.py#L172 and the following `optimize` method. Are there any patterns that you see that should still be removed?