florianfesti/boxes #64: Running setup.py does not work as expected

nobels, 2018-04-14:
Hello, 

I would like to generate this file @ inkex folder but it's still empty. 

My environment is windows 10 with Python 3.6.3

This i my output
```
C:\Projects\Fablab\boxes>py setup.py build
running build
running build_py
buildInkscapeExt()
'scripts\boxes2inkscape' is not recognized as an internal or external command,
operable program or batch file.
running build_scripts
```

Nice job on what you build there ;-)

florianfesti, 2018-04-15:
Looks like the problem is that scripts\boxes2inkscape is not executable under Windows. Can you try renaming it to scripts\boxes2inkscape.py and adjust the name in buildInkscapeExt() in the setup.py file, please. This will still require *.py files being registered with Windows, I guess.

I do not use Windows at all and it shows in things like this.

florianfesti, 2018-04-15:
Ok, after thinking a bit just calling Python directly may be the easier and more robust solution.

commit 0a25d8dc (2018-04-15): Setup.py: Call python interpreter directly to avoid issues on Windows

Fixes #64
closed 2018-04-15 by florianfesti