florianfesti/boxes #66: Add settings to SVG comments

bwente, 2018-05-17:
I found that I needed to redo my file and could not remember the settings. I just needed to change one part but I had to remember all the settings. It would be very helpful to have the settings embeded in the file.

```
<!-- 
https://www.festi.info/boxes.py/RegularBox?FingerJoint_angle=90.0&FingerJoint_surroundingspaces=2.0&FingerJoint_edge_width=1.0&FingerJoint_finger=2.0&FingerJoint_play=0.0&FingerJoint_space=2.0&FingerJoint_width=1.0&h=100.0&outside=0&outside=1&radius=50.0&n=4&top=none&thickness=3.0&format=svg&tabs=0.0&debug=0&reference=100&burn=0.1&render=1
-->
```

Rotzbua, 2018-08-06:
To make this simple I created a greasemonkey script:
https://gist.github.com/Rotzbua/691361a41722fb0ef6a25ec4a42a12f4

Install:
https://gist.github.com/Rotzbua/691361a41722fb0ef6a25ec4a42a12f4/raw/greasemonkey.festi.info.user.js

bwente, 2018-08-07:
Perfect! Very elegant workaround. I now learned something new (Tampermonkey).

bwente, 2018-08-07:
I was hoping I could change the url parameter "render" to 0 and get the setting to restore the settings in the form.

florianfesti, 2018-08-09:
I'll have a look into that. But it may take a while.

florianfesti, 2018-08-14:
Ok, fixed. Removing the render param or passing 0 instead of 1 now opens the input form filled with the values given in the URL params. @Rotzbua You may want to strip the `&render=1` from the end of the URL in your greasemonkey script.

bwente, 2018-08-14:
This is so great! Thank you! What a time saver, and a great way to share and post box "recipes". I can prototype a design in 5mm cardboard and then go back and export it back out for 3mm hardwood.

florianfesti, 2018-08-14:
@bwente @Rotzbua  I mentioned you guys at https://hackaday.io/project/10649/log/150958-reuse-your-previous-box-settings 
Drop me a note if you want to be referred to with another account (e.g. hackaday.io).

Rotzbua, 2018-08-15:
Thanks for implementation :+1: 

I found a bug with checkboxes. Checkboxes are checked but it is `0` e.g. `&debug=0`. 

### Reason

Doubled `input` tag.

```html
<td>debug</td>
<td>
 <input name="debug" type="hidden" value="0">
 <input name="debug" type="checkbox" value="1" checked="checked">
</td>
```
### Reproduce

Checkbox not checked:
https://festi.info/boxes.py/ElectronicsBox

Checkbox checked as expected:
https://festi.info/boxes.py/ElectronicsBox?debug=1

Bug: checkbox checked but `0`:
https://festi.info/boxes.py/ElectronicsBox?debug=0

florianfesti, 2018-08-16:
Thanks for testing! Fixed!

florianfesti, 2019-07-04:
Ok, now the svg file is completely parsed back in using eTree. This allows to finally add this directly. Still need to figure out what exactly to add. Inkscape is using <metadata><RDF><Work>....</ May be that' the way to go...?

bwente, 2019-07-05:
Adobe simply uses comments...

``

florianfesti, 2019-07-06:
*Urgs* no -- in xml comments. So I guess this has to do without the command line...

commit fff74cf3 (2019-07-07): Add metadata as a xml comment

Resolves #66
closed 2019-07-07 by florianfesti