florianfesti/boxes #70: Anyway to get rid of these rounded corners?

nach00, 2018-07-01:
Hi

Please see the following image:
https://imgur.com/a/cmucwkG

These corners are compiled in boxes.py. They are of the holes for the bintray. I would like to use the Multicam CNC machine to cut out of wood, so I will not be using laser. These lines interfere with the path processing of Vcarve, the software used to program the paths for the machine. It is because the lines intersect and the software does not like that.

Right now, I have to manually snip each corner of each hole.

The reason I need these cut is because the cut width of the bit is much larger than the laser, so I need to add "dog bones" to the corners because the drill bit is round and it cannot cut a sharp right angle. I am using a 1/8" bit.

Please let me know if there is a setting to remove this rounded corner. Thank you.

florianfesti, 2018-07-02:
You can try using a burn value of zero. This should solve the overlapping lines. But then you need to do the tools compensation on your own.

Note that the current path kinda does dog holes already - just not very nice ones.

The underlying problem is that Boxes.py does tool compensation (aka burn) by keeping the length of straight lines and adjusting the radii of the curves. The artifact you see above is a radius "of negative length" that connects the two  straight lines. To get rid of this Boxes.py would need to shorten the straight lines. But this is difficult to do given the way it works right now as the straight lines are oblivious about what is following them.

nach00, 2018-07-02:
ok thanks! I'll give it a shot and let you know how it turns out.

florianfesti, 2018-07-23:
Did that help?

nach00, 2018-07-25:
Hello,

Yes, eliminating the burn value removed the corner rounds completely. This was perfect for what I needed.

I used Adobe Illustrator to combine all the vectors together with a paid plugin called VectorFirstAid by Astute Graphics. When I open the generated file in AI, all the segments of the boxes are separate vectors, so there can be thousands of individual vectors depending on the file created. VectorFirstAid provides a simple one click solution to joining all the vectors.

Then, I needed to create two sets of cut paths. One for an inside cut (the holes) and one for the outside cut (the outside border). I measured the thickness of my material with a digital caliper and use this value as the thickness in boxes.py. 

I was pretty close to getting it right. My mistake was that I didn't try to sample a small 1x1x1 box to get the values perfect before attempting a multi-compartment box. I messed up a few times (first by not splitting the inside and outside path) and also by using the given thickness value of the wood (I bought 3/4" plywood, it really came out to like .7" or something). I really felt like I had all the settings done perfectly. I properly mounted my material to the cutting area, but during the cut, the machine finished an outline section of my cut. I knew my wood was a little warped (it had been sitting vertically for a few days) which is why I tried to mount it perfectly, but I think my nesting of the pieces were way too tight. When the wood released from the cut (I even used tabs to hold them down, but the machine wasn't 100% properly calibrated, it's a makerspace tool so it gets used and abused), the wood bent upward, and as the tool moved to its next location, it ran into the bent wood and snapped the drill bit.

I had been working a few days to figure it out, and after being out of material and the lumber yard not open, I decided to give up.

Next time I'm probably going to use MDF or hardboard, probably 1/8" or 1/4" thick. I will update again once I'm successful with my settings.

florianfesti, 2018-07-25:
OK, thanks for the update. I'll close the ticket as there is probably nothing that can be done right here. Feel free to post another update here anyway.

closed 2018-07-25 by florianfesti