florianfesti/boxes #72: Way to force viewBox Y origin to 0?

dwardio, 2018-07-18:
The default viewBox origin (9000+ in every generated file so far) prevents the SVG files from loading properly in Affinity Designer on a Mac.  I can manually edit each file to set the viewBox origin to 0,0 and they work fine, but it would be great to know (1) why it's set so high by default, and (2) if there's a way to force it to 0?

florianfesti, 2018-07-19:
Not that easily. The reason for starting drawing at Y 10000 is that 
a) The graphics library I use (cairo) requires to determine the size of the drawing in advance. So I have to choose something that is big enough - like 10000mm.
b) The direction of the Y axis is top to bottom in SVG while I draw from bottom to top. SO I have to start at the high Y numbers.

After the drawing is finished I reduce the size of the canvas in the SVG file by changing some numbers. As far a I know the resulting SVG should be legal and your software not being able to read it is probably a bug - or misguided plausibility check.

At some point in the future I may reread and optimize the whole SVG file. But that'd require a lot of new code and probably also a lot of RAM and computation time.

florianfesti, 2018-09-08:
As this may take a while to get fixed in Boxes.py you should also open a ticket against Affinity Designer. While I doubt they are much faster in fixing this they are the ones at fault here. At least as I understand the SVG standard.

florianfesti, 2019-07-04:
Ok, this is now done with 31d46088c21bb5664bd1282df7d11e93e62b3dbb. Should hit the web instance the next few days.

closed 2019-07-04 by florianfesti