florianfesti/boxes #73: Add counter bedbolts

miguelmorin, 2018-07-24:
Slide 10 of `presentation.odp` shows a box with bedbolts. If the box is large and the pieces to the right are top and bottom, then they serve as anchors for the nut of the bedbolt, but the bottom piece could fall off. I suggest to put holes for screws to attach these pieces to the other ones, so the side pieces would have both nuts and screws of the bedbolts.

florianfesti, 2018-07-24:
That's not quite how the box goes together. The piece at the top is the floor of the box. It's fingers go into the slots of the 4 side walls and keeps them together. The vertical edges are held together with bolts, too. The go through the holes on the fingers of the left walls and into the screw slots of the walls on the right.

Looks like there is still a bug in the drawing as the side walls have 3 screw slots when there are only 2 holes on the other sides - probably passing the wrong (bottom) settings.

As the slides are pretty old I am closing this ticket for now. Feel free to reopen as soon as there is actually some faulty implementation exposed.

closed 2018-07-24 by florianfesti

miguelmorin, 2018-07-24:
Right. I want to build a mailbox which opens at the front, so I saw Box2.py with a front opening. I will try UniversalBox as suggested in #36 and play a little with the bedbolts to see if they hold fast with 5mm acrylic.

miguelmorin, 2018-07-27:
I adapted `closedbox.py` to include counter-bolts near the bolts, with the result below. I will test both tomorrow in acrylic with the laser cutter and report on the merits of each.
[image]