florianfesti/boxes #74: Error in installation: could not create '/usr/share/inkscape'

miguelmorin, 2018-07-24:
MacOS now has [System Integrity Protection](https://stackoverflow.com/questions/33309005/sudo-mkdir-in-usr-operation-not-permitted-el-capitan), which prevents writing to `/usr/` unless it's in `/usr/local/`. Installing `Boxes` fails:

```
> python3 setup.py install
...
creating /usr/share/inkscape
error: could not create '/usr/share/inkscape': Operation not permitted
```

I am running macOS El Capitan 10.11.6. Creating that directory also fails in High Sierra 10.13.6.

florianfesti, 2018-07-24:
setup.py tries to install the inkscape extensions to  /usr/share/inkscape/extensions/
You can adjust the path in there.  If you post the proper path here I can add in if clause for OS X.

May be I should make the inkspace extensions optional...

miguelmorin, 2018-07-24:
I confirm that using the path `"/usr/local/share/inkscape/extensions/"` works and that the path `"/usr/share/inkscape/extensions/"` does not. Shall I do a pull request with a branch for before and after El Capitan?

florianfesti, 2018-07-24:
Both fine with me. I am going to <rebase and merge> anyway.

florianfesti, 2018-07-24:
Ok, should be fixed. Please reopen if you encounter issues.

closed 2018-07-24 by florianfesti

miguelmorin, 2018-07-25:
I confirm that the latest commit fixes the issue. Thanks!