florianfesti/boxes #75: boxes: error: argument --[some argument]: expected one argument

miguelmorin, 2018-07-24:
I installed `cffi`, `cairocffi`, and `boxes`. I attempt to generate a box with

```
boxes universalbox --thickness 5 --output ~/test.svg --format svg --x 254 --y 300 --h 100 --lid flat
```
and get the error
```
boxes: error: argument --lid: expected one argument
```

If I remove the last argument, I get a similar error:
```
boxes: error: argument --h: expected one argument
```

I could not find the right syntax in the README nor in the issues. How can I call `boxes` with a generator in the local system (whose source code I will modify in the future)?

florianfesti, 2018-07-24:
Boxes.py uses the argparse library. It requires a = connection long arguments and the value. So

`
boxes universalbox --thickness=5 --output=test.svg --format=svg --x=254 --y=300 --h=100 --lid=flat
`
You can also just start boxes/boxesserver and use the web interface at localhost:8000

closed 2018-07-24 by florianfesti

miguelmorin, 2018-07-25:
Yes, this command works. Thanks!