florianfesti/boxes #76: Inkscape plugin not working?

speendo, 2018-07-31:
First of all thank you for boxes.py.

I tried to gain a first insight to this project by using the inkscape plugin on my Debian Testing x64 system with gnome.

I installed boxes.py by cloning the github project and then running setup.py as sudo user. The Inkscape plugin has been installed, however it does not work. All the actual functions are missing. See the screenshot.

I hope I didn't make a mistake!?

Thank you in advance!

[image]

florianfesti, 2018-08-03:
This is probably caused by Inkscape not finding the `boxes` script. Can you have a look into one of the .inx files what it tries to execute and then look were the `boxes` script ended up.

speendo, 2018-08-14:
Hm, this is what `boxes.py.WineRack.inx` has inside:

    <?xml version="1.0" encoding="UTF-8"?>
    <inkscape-extension xmlns="http://www.inkscape.org/namespace/inkscape/extension">
      <_name>WineRack</_name>
      <dependency type="executable" location="path">boxes</dependency>
      <id>info.festi.boxes.py.WineRack</id>

      <param name="generator" type="string" gui-hidden="true">winerack</param>
      <param name="tab" type="notebook">

        <page name="0" _gui-text="WineRack Settings">
          <param name="x" type="float" max="9999" _gui-text="x" gui-description="inner width in mm (unless outside selected)">400</param>
          <param name="y" type="float" max="9999" _gui-text="y" gui-description="inner depth in mm (unless outside selected)">300</param>
          <param name="h" type="float" max="9999" _gui-text="h" gui-description="inner height in mm (unless outside selected)">210</param>
          <param name="radius" type="float" max="9999" _gui-text="radius" gui-description="Radius of comb">46.0</param>
          <param name="walls" type="enum" _gui-text="walls" gui-description="which of the honey comb walls to add">
            <item value="minimal">minimal</item>
            <item value="no_verticals">no_verticals</item>
            <item value="all" selected="selected">all</item>
          </param>
        </page>

        <page name="1" _gui-text="Default Settings">
          <param name="thickness" type="float" max="9999" _gui-text="thickness" gui-description="thickness of the material">3.0</param>
          <param name="tabs" type="float" max="9999" _gui-text="tabs" gui-description="width of tabs holding th parts in place (not supported everywhere)">0.0</param>
          <param name="debug" type="boolean" max="9999" _gui-text="debug" gui-description="print surrounding boxes for some structures">False</param>
          <param name="reference" type="float" max="9999" _gui-text="reference" gui-description="print reference rectangle with given length">100</param>
          <param name="burn" type="float" max="9999" _gui-text="burn" gui-description="burn correction in mm (bigger values for tighter fit)">0.1</param>
        </page>

        <page name="2" _gui-text="Settings for Finger Joints">
          <param name="FingerJoint_angle" type="float" max="9999" _gui-text="angle" gui-description="">90.0</param>
          <param name="FingerJoint_surroundingspaces" type="float" max="9999" _gui-text="surroundingspaces" gui-description="maximum space at the start and end in multiple of normal spaces">2.0</param>
          <param name="FingerJoint_edge_width" type="float" max="9999" _gui-text="edge_width" gui-description="space below holes of FingerHoleEdge">1.0</param>
          <param name="FingerJoint_finger" type="float" max="9999" _gui-text="finger" gui-description="width of the fingers">2.0</param>
          <param name="FingerJoint_play" type="float" max="9999" _gui-text="play" gui-description="extra space to allow movement">0.0</param>
          <param name="FingerJoint_space" type="float" max="9999" _gui-text="space" gui-description="space between fingers">2.0</param>
          <param name="FingerJoint_width" type="float" max="9999" _gui-text="width" gui-description="width of finger holes">1.0</param>
        </page>

      </param>
      <effect>
        <object-type>all</object-type>
          <effects-menu>
            <submenu _name="Boxes.py">
             <submenu _name="Shelves"/>
            </submenu>
          </effects-menu>
      </effect>
      <script>
        <command reldir="extensions">boxes</command>
      </script>
    </inkscape-extension>

So, it seems to try to run `boxes` right? This should work in principle. When I run `boxes` from the commandline I get

    $ boxes
    boxes.py

    Generate stencils for wooden boxes.

    Usage:
      boxes <generator> [<args>...]
      boxes --list
      boxes (-h | --help)
      boxes --version

    Options:
      -h --help     Show this screen.
      --version     Show version.
      --list        List available generators.

speendo, 2018-08-14:
I was able to fix it. I thought it's enough that `boxes` is in the path. Turned out it needs to be symlinked to the extensions directory (you mentioned that in the docs anyway, for some reason I thought that wouldn't apply for Linux systems).

However, I also needed to install the package `python3-lxml`. I think this requirement is not in the docs yet...

Thank you anyway for your help and these great scripts and sorry for bothering.

speendo, 2018-08-15:
lxml is in the docs. Sorry, must be the heat ...

florianfesti, 2018-08-16:
Don't get me started about the heat...

Anyway, ... I guess we are done here. Feel free to reopen or open a new ticket if you run into more problems. Closing.

closed 2018-08-16 by florianfesti