florianfesti/boxes #77: Create new boxes

nobels, 2018-08-16:
Hey i like your project but i have a quetion:

I would like to create new boxes -> differnt from the one's you have ...

How do you start working on this ? 

Any tips & tricks you can share ?

Thanks !

florianfesti, 2018-08-16:
There is a skeleton for new boxes in boxes/generators/_template.py
Copy it to start your new generator. There are comments on what you need to change. You can delete some of the example code in there, if you don't need the stuff it is doing.
Have a look at some other generators.
The lib is divided in different level of abstractions. It is hard to tell you where to look without knowing what you want to do.
There is an (slightly outdated) API doc linked at the bottom of the web interface.
You need to understand the Boxes.rectangularWall() method and the following of its params
- x, y obviously...
- edges
- callback
- move
They are actually described in the docs: http://florianfesti.github.io/boxes/html/Boxes.html#generating-parts

You may want to create your own parts. May be you can get away with "just" implementing your own edge type. Or implement callbacks to create the holes you need in the parts.

florianfesti, 2018-08-16:
Oh, run scripts/boxesserver for development. It will restart itself on changes. But may not pick up new files or choke on broken syntax.

nobels, 2018-08-25:
Sorry but how do you start boxesserver ?
i use this command 'python boxesserver' but does not work. :-(

fyi
I have python 3 installed.

florianfesti, 2018-08-26:
Hmm, I haven't used Windows since I deleted my Windows NT partition...
May be 460691f0aa7ed5085a908450e076ce03fb535051 helps.
What error message do you get?

florianfesti, 2018-08-27:
I updated the docs a bit. But there is still a lot more to do in this area.