florianfesti/boxes #78: Build failure at current head

greenaar, 2018-08-21:
07:11:53 buildInkscapeExt()
07:11:54 Traceback (most recent call last):
07:11:54   File "setup.py", line 57, in <module>
07:11:54     keywords=["boxes", "box", "generator", "svg", "laser cutter"], )
07:11:54   File "/opt/boxes/lib/python3.5/site-packages/setuptools/__init__.py", line 131, in setup
07:11:54     return distutils.core.setup(**attrs)
07:11:54   File "/usr/lib/python3.5/distutils/core.py", line 148, in setup
07:11:54     dist.run_commands()
07:11:54   File "/usr/lib/python3.5/distutils/dist.py", line 955, in run_commands
07:11:54     self.run_command(cmd)
07:11:54   File "/usr/lib/python3.5/distutils/dist.py", line 974, in run_command
07:11:54     cmd_obj.run()
07:11:54   File "/usr/lib/python3.5/distutils/command/build.py", line 135, in run
07:11:54     self.run_command(cmd_name)
07:11:54   File "/usr/lib/python3.5/distutils/cmd.py", line 313, in run_command
07:11:54     self.distribution.run_command(command)
07:11:54   File "/usr/lib/python3.5/distutils/dist.py", line 974, in run_command
07:11:54     cmd_obj.run()
07:11:54   File "setup.py", line 30, in run
07:11:54     self.distribution.data_files["/usr/share/inkscape/extensions/"].append('scripts/boxes')
07:11:54 TypeError: list indices must be integers or slices, not str

Was building correctly until about a week ago.

greenaar, 2018-08-22:
Thanks! That *does* fix that issue, though now cairocffi needs to be newer than what distributes with ubuntu 16.04 (0.8 minimum) -- works with ubuntu 18.04, or with that package installed via pip.

The next issue occurs later:

19:51:10 copying inkex/boxes.py.AngledCutJig.inx -> /usr/share/inkscape/extensions/
19:51:10 error: can't copy 's': doesn't exist or not a regular file

florianfesti, 2018-08-22:
Hmmm, that's actually supposed to work...
What exact command are you executing?
Can you please check that inkex/boxes.py.AngledCutJig.inx and /usr/share/inkscape/extensions/ actually exists? Thanks!

greenaar, 2018-08-22:
Build output:

08:11:46 Step 7/12 : RUN virtualenv /opt/boxes -p /usr/bin/python3 --system-site-packages
08:11:46  ---> Using cache
08:11:46  ---> e167d1134538
08:11:46 Step 8/12 : COPY boxes/. .
08:11:48  ---> 1ce6e90d61c4
08:11:48 Removing intermediate container 1789e17abf88
08:11:48 Step 9/12 : RUN /opt/boxes/bin/python setup.py build &&     /opt/boxes/bin/python setup.py test &&     /opt/boxes/bin/python setup.py install
08:11:48  ---> Running in 60b8998ca7f5
08:11:51 running build
08:11:51 running build_py
08:11:51 buildInkscapeExt()
08:11:52 creating build
08:11:52 creating build/lib
08:11:52 creating build/lib/boxes
08:11:52 copying boxes/pulley.py -> build/lib/boxes
08:11:52 copying boxes/servos.py -> build/lib/boxes
08:11:52 copying boxes/robot.py -> build/lib/boxes
08:11:52 copying boxes/__init__.py -> build/lib/boxes
08:11:52 copying boxes/parts.py -> build/lib/boxes
08:11:52 copying boxes/gears.py -> build/lib/boxes
08:11:52 copying boxes/formats.py -> build/lib/boxes
08:11:52 copying boxes/svgutil.py -> build/lib/boxes
08:11:52 copying boxes/vectors.py -> build/lib/boxes
08:11:52 copying boxes/lids.py -> build/lib/boxes
08:11:52 copying boxes/edges.py -> build/lib/boxes
08:11:52 creating build/lib/boxes/generators
08:11:52 copying boxes/generators/planetary2.py -> build/lib/boxes/generators
08:11:52 copying boxes/generators/pulley.py -> build/lib/boxes/generators
08:11:52 copying boxes/generators/angledbox.py -> build/lib/boxes/generators
08:11:52 copying boxes/generators/organpipe.py -> build/lib/boxes/generators
08:11:52 copying boxes/generators/hingebox.py -> build/lib/boxes/generators
08:11:52 copying boxes/generators/folder.py -> build/lib/boxes/generators
08:11:52 copying boxes/generators/display.py -> build/lib/boxes/generators
08:11:52 copying boxes/generators/ubox.py -> build/lib/boxes/generators
08:11:52 copying boxes/generators/rack19box.py -> build/lib/boxes/generators
08:11:52 copying boxes/generators/ottolegs.py -> build/lib/boxes/generators
08:11:52 copying boxes/generators/robotarm.py -> build/lib/boxes/generators
08:11:52 copying boxes/generators/_template.py -> build/lib/boxes/generators
08:11:52 copying boxes/generators/silverwarebox.py -> build/lib/boxes/generators
08:11:52 copying boxes/generators/drillbox.py -> build/lib/boxes/generators
08:11:52 copying boxes/generators/traylayout.py -> build/lib/boxes/generators
08:11:52 copying boxes/generators/hooks.py -> build/lib/boxes/generators
08:11:52 copying boxes/generators/waivyknob.py -> build/lib/boxes/generators
08:11:52 copying boxes/generators/flexbox3.py -> build/lib/boxes/generators
08:11:52 copying boxes/generators/__init__.py -> build/lib/boxes/generators
08:11:52 copying boxes/generators/nemamount.py -> build/lib/boxes/generators
08:11:52 copying boxes/generators/openbox.py -> build/lib/boxes/generators
08:11:52 copying boxes/generators/printer.py -> build/lib/boxes/generators
08:11:52 copying boxes/generators/concaveknob.py -> build/lib/boxes/generators
08:11:52 copying boxes/generators/flexbox.py -> build/lib/boxes/generators
08:11:52 copying boxes/generators/flextest.py -> build/lib/boxes/generators
08:11:52 copying boxes/generators/ottobody.py -> build/lib/boxes/generators
08:11:52 copying boxes/generators/gearbox.py -> build/lib/boxes/generators
08:11:52 copying boxes/generators/magazinefile.py -> build/lib/boxes/generators
08:11:52 copying boxes/generators/castle.py -> build/lib/boxes/generators
08:11:52 copying boxes/generators/typetray.py -> build/lib/boxes/generators
08:11:52 copying boxes/generators/planetary.py -> build/lib/boxes/generators
08:11:52 copying boxes/generators/roundedbox.py -> build/lib/boxes/generators
08:11:52 copying boxes/generators/gear.py -> build/lib/boxes/generators
08:11:52 copying boxes/generators/universalbox.py -> build/lib/boxes/generators
08:11:52 copying boxes/generators/rack10box.py -> build/lib/boxes/generators
08:11:52 copying boxes/generators/closedbox.py -> build/lib/boxes/generators
08:11:52 copying boxes/generators/bintray.py -> build/lib/boxes/generators
08:11:52 copying boxes/generators/trayinsert.py -> build/lib/boxes/generators
08:11:52 copying boxes/generators/trafficlight.py -> build/lib/boxes/generators
08:11:52 copying boxes/generators/arcade.py -> build/lib/boxes/generators
08:11:52 copying boxes/generators/electronicsbox.py -> build/lib/boxes/generators
08:11:52 copying boxes/generators/displayshelf.py -> build/lib/boxes/generators
[... 246 lines cut ...]
08:11:59 copying inkex/boxes.py.DisplayShelf.inx -> /usr/share/inkscape/extensions/
08:11:59 copying inkex/boxes.py.Planetary.inx -> /usr/share/inkscape/extensions/
08:11:59 copying inkex/boxes.py.PoleHook.inx -> /usr/share/inkscape/extensions/
08:11:59 copying inkex/boxes.py.UnevenHeightBox.inx -> /usr/share/inkscape/extensions/
08:11:59 copying inkex/boxes.py.OttoLegs.inx -> /usr/share/inkscape/extensions/
08:11:59 copying inkex/boxes.py.FlexBox4.inx -> /usr/share/inkscape/extensions/
08:11:59 copying inkex/boxes.py.FlexTest2.inx -> /usr/share/inkscape/extensions/
08:11:59 copying inkex/boxes.py.OttoBody.inx -> /usr/share/inkscape/extensions/
08:11:59 copying inkex/boxes.py.Arcade.inx -> /usr/share/inkscape/extensions/
08:11:59 copying inkex/boxes.py.TypeTray.inx -> /usr/share/inkscape/extensions/
08:11:59 copying inkex/boxes.py.OpenBox.inx -> /usr/share/inkscape/extensions/
08:11:59 copying inkex/boxes.py.WineRack.inx -> /usr/share/inkscape/extensions/
[... cut ...]
08:11:59 copying inkex/boxes.py.Silverware.inx -> /usr/share/inkscape/extensions/
08:11:59 copying inkex/boxes.py.AngledCutJig.inx -> /usr/share/inkscape/extensions/
08:11:59 error: can't copy 's': doesn't exist or not a regular file
08:12:00 The command '/bin/sh -c /opt/boxes/bin/python setup.py build &&     /opt/boxes/bin/python setup.py test &&     /opt/boxes/bin/python setup.py install' returned a non-zero code: 1
08:12:00 Build step 'Execute shell' marked build as failure
08:12:00 Finished: FAILURE

florianfesti, 2018-08-22:
Looks like the fix was just not good enough. 's' is probably the first letter of `scripts/boxes`. Let's hope that wrapping the string into a list fixes that. For some reason I had trouble seeing the same error locally - although it really should be there.
Guess it shows a bit that I only run Boxes.py directly from the checkout...

greenaar, 2018-08-22:
Yep. that did it.  This application continues to improve, so I appreciate the opportunity to keep up with the updates.

Thanks, that was a speedy set of fixes. Closing this ticket.

closed 2018-08-22 by greenaar