florianfesti/boxes #83: Installation Docs for Windows

florianfesti, 2018-09-08:
Setting up Boxes.py on Windows is not entirely trivial. Main issue is getting the cairo library working. Things are even more complicated if things are expected to work with Inkscape. There are various bits of information distributed in different tickets but no clear and tested instructions.

If some actually using Windows and got Boxes.py running (may be even with Inkscape and every bit helps) could provide instructions that be great.

florianfesti, 2018-09-08:
Oh, as already promised else where: I offer writing a (not too complicated) generator for the one providing the docs.

Rotzbua, 2018-10-01:
## Install under windows

I tried to install under win8.1 with python3.6 32bit. No success so far.

### Results

No official windows support for cairo, according to issue:
https://github.com/pygobject/pycairo/issues/19

Unofficial package:
https://www.lfd.uci.edu/~gohlke/pythonlibs/#pycairo

Package name:

> pycairo‑1.17.1‑cp36‑cp36m‑win32.whl

cp36 --> python 3.6
win32 --> python 32bit version

Install:

> pip install pycairo-1.17.1-cp36-cp36m-win32.whl

Works. Then boxes setup script. Seems to work.

Then run:

> python scripts/boxesserver

Error:

>   File "XXX\Python36-32\lib\site-packages
> \cairocffi-0.8.0-py3.6.egg\cairocffi\__init__.py", line 41, in <module>
>     cairo = dlopen(ffi, 'cairo', 'cairo-2')
>   File "XXX\Python36-32\lib\site-packages
> \cairocffi-0.8.0-py3.6.egg\cairocffi\__init__.py", line 38, in dlopen
>     raise OSError("dlopen() failed to load a library: %s" % ' / '.join(names))
> OSError: dlopen() failed to load a library: cairo / cairo-2

Well.. back to linux vmware..

Rotzbua, 2018-10-03:
> 
> 
> Oh, as already promised else where: I offer writing a (not too complicated) generator for the one providing the docs.

A travis script makes life easy, see #90

florianfesti, 2018-11-15:
These inofficial pycairo wheels at least contain a cairo.dll:  
https://www.lfd.uci.edu/~gohlke/pythonlibs/#pycairo
May be they can do the trick.

commit 9135486d (2019-02-26): Add installation guide for Windows (fixes #83)
closed 2019-02-26 by florianfesti