florianfesti/boxes #103: Ponoko-ready SVG generator

wayn3w, 2018-12-04:
Hello folks,

I'd like to suggest a small feature to the SVG generation:  to make a 'Ponoko'-ready' SVG file, one that can be uploaded to Ponoko for cutting without having to go through Inkscape or some other SVG editor to make changes.  They have specific requirements which could affect the cutting the tolerances.

The setting would be a boolean (default off) **--for-service** which would do the following initially:
* make all lines blue
* make the width of the cutting line 0.01mm
* add Inkscape-styled comments and/or metadata that indicate it came from boxes.py version yada yada with parameters yada yada yada.  The comments would be in a different layer initially locked.

Future enhancements could be
* make it fit into a specific Ponoko template (P1,P2,P3) either by autofit, or by user selection

I'm happy to do the work because I've been to do meaning SVG-based development for some time.  Please let me know if this is acceptible.  Thanks,  
Wayne

florianfesti, 2018-12-04:
I would probably not add a switch but a new output --format "SVG_Ponko".
Currently the use of colour is a bit adhoc. But there are only few places in the code that need changes. May be a bit of refactoring would not hurt that code.
Setting a different line width should be pretty easy.
SVG style comments is probably difficult as Boxes.py is based on cairo. As a general graphics library it does not really support (many) output specific options AFAIK. But if someone can figure out how to do this this is clearly on my wish list.

So yes, patches are welcome.

florianfesti, 2018-12-04:
See #66 wrt SVG comments.

wayn3w, 2018-12-05:
Thank you.  I'll make it the new format, as you suggested, and I'll start
with refactoring.

wayn3w, 2018-12-12:
I added a pull-request for this work:  https://github.com/florianfesti/boxes/pull/105

florianfesti, 2019-02-26:
Finally updated the web instance. Thanks to @wayn3w for all the work!

closed 2019-02-26 by florianfesti