florianfesti/boxes #107: TypeTray should have a top edge param

florianfesti, 2019-01-02:
TypeTray only has two options for the top: Straight edge or closed top (using FingerJointEdge f).

This leaves at least the StackableEdge available for the bottom unmatched. One could argue the top edge should actually be freely selectable similar to the UniversalBox. Otoh this will make the UniversalBox redundant - which may be a good thing...

See also https://hackaday.io/project/10649-boxespy/discussion-118236

Blancanegra, 2019-01-08:
Please keep UniversalBox, I found it very useful!

Maybe you should add inner divisions to UniversalBox, or lid support for TypeTray... crossing worlds! :)

florianfesti, 2019-01-08:
I already added a top_edge param to the TypeTray  (https://github.com/florianfesti/boxes/commit/c0a782b745cb51909c6560fd9ec3a58ec88d200e). But the change is not in the web instance yet.
I will probably keep the UniversalBox to not confuse users too much, although the TypeTray can (almost) do everything the UniversalBox can plus the inner walls.

florianfesti, 2019-01-10:
Ok, I was still missing the lids that belong to the top_edges. That's fixed now and already online. I will probably change the sides of the outside walls to the same f-F pattern as the universal box. But that's out of scope of this ticket. Closing as fixed.

closed 2019-01-10 by florianfesti