florianfesti/boxes #108: Paint Storage - customizable grip hole

Blancanegra, 2019-01-08:
The default grip hole width makes the walls too fragile. For small organizers a hole in the wall to pass a hand finger would be enough to catch and carry it safely, keeping the wall strong.

Please consider to add "gripheight" and "gripwidth"parameters (like in type tray) to the generator.

florianfesti, 2019-01-08:
I'd rather have a sane default.
https://github.com/florianfesti/boxes/blob/master/boxes/generators/paintbox.py#L79 creates the hole with h/4., 3./4.*x being the size. Can you try and then suggest some smaller values?

Blancanegra, 2019-01-09:
I'm using the web generators on windows so I can't try the code but sure 3./4,*x is too wide.

In my case I have a lot of cans from common paints in the market: Tamiya, Americana, Vallejo... for my 3 mm thick shelfs a 16 mm diameter  hole seems to be enough to pass trough a finger and carry the stacked shelfs safely.

[image]
My shelfs, with no hole. 

Maybe a sane rule would consider ergonomic data (min & max values) like finger diameter and hand with to calculate proper holes. :)

florianfesti, 2019-01-10:
Reduced width of the hole to half of the shelf. That should double the strength. I also increased the radius to max.

florianfesti, 2019-01-10:
This ticket has a better photo than some Boxes.py generators...

commit 371097aa (2019-01-10): PaintStorage: Make grip hole smaller and more rounded

Fixes #108
closed 2019-01-10 by florianfesti