florianfesti/boxes #110: Feature Request: Stackable Storage Bins

Christian308, 2019-01-16:
Thanks for sharing such a great software.

I am wondering if it is possible to generate stackable storage bins such as
https://www.thingiverse.com/thing:1279971
with it. Those style of boxes (usually made out of plastic, https://www.linbins.com/products/linbin-trays-bins/linbins) are a staple in any workshop.

Important features:
- Stackable, they interlock so they do not easily slide off each other
- Lower part of front wall is angled so small parts can be easier retrieved from bin 
- Middle part of front wall is straight to attach a label
- Front wall does not go all the way to the top to be able to see and retrieve items from stacked bin.

florianfesti, 2019-01-16:
Yeah, I have been thinking about something like this  a few weeks ago. Still I don't have a good way of doing them yet. I have doubts that those little lugs will really hold the bin on top of each other securely. The plastic ones are covered in ribs and ridges. The bottom should probably be held with a finger hole edge. If you can come up with a sketch I am happy to do the programming.

florianfesti, 2019-05-11:
Guess this is not happening any time soon. Closing

closed 2019-05-11 by florianfesti

Christian308, 2019-05-13:
Sorry for the late response. Sometimes life gets in the way. I tested the original design and the small lugs hold the bins in place quite securely. See attached video. 
[StackableStorageBinsDemoSmall.zip](https://github.com/florianfesti/boxes/files/3174920/StackableStorageBinsDemoSmall.zip)