florianfesti/boxes #111: Stackable bottom edge with finger joints

Blancanegra, 2019-01-18:
Actually there are no finger joints along the bottom edges "s" and "h"" below holes of FingerHoleEdge.
Extending the edge type to the end would make the legs stronger even for low "s" and "h" edge heights.

[image]
Top figure: actual stackable edge.
Bottom figure: proposed edge fix.

florianfesti, 2019-01-19:
This is a bit more tricky than it looks.
I have to think a bit to figure out how to do this. The issue here is that the height of the wall is distance from the top edge to the top of the finger holes. This way a adjacent wall of the same height fits there if it has a "f" edge on the side - no matter what top or bottom edge it has. Extending the side edge down to the floor requires adjusting some sizes and breaks the compatibility between all bottom edge types (F, h, s).
This can be done for some generators but not for arbitrary walls.

florianfesti, 2019-12-14:
Ok, turns out adding `ignore_widths=[1, 6]` to the side walls does just this. Now I just need to modify all generators that make use if h and s edges or the bottom_edge param...

closed 2019-12-14 by florianfesti

Blancanegra, 2019-12-14:
Great!
Thank you.

commit aba9631b (2019-12-14): Make joints of box sides go to the very bottom

for NotesHolder, TypeTray and UniversalBox.
The same should be done to many other boxes as soon as they get the
bottom_edge parameter

Resolves: #111