florianfesti/boxes #114: reference and svg_ponoko

wayn3w, 2019-02-04:
The reference is problematic with svg_ponoko, but the rest seems fine.  There's a couple of things here, and I hesitate to offer a fix because it seems there is some history in that code:

* WHITE is not a valid color for use by Ponoko.  If that rectangle is meant to be seen, for Ponoko it should be red (for  etching).  That rectangle also seems to intersect the last letter on a Mac.
* The large rectangle should be red (etching), or blue (cutting).  Black is for raster print.
* The letters are okay if one wants to raster print them.  It would be cheaper to give an option to omit the fill, but put a red stroke on the text (0.01mm) so it will be etched.
* The last argument to text(), color, should have a default of Color.BLACK, not [0, 0, 0]

I'm happy to make appropriate changes to support a Ponoko design, but could you tell me why the rectangle is white?

Should saved_context() be used here instead of the @restore?
Does the last two lines need to be there if the context is restored; that is the set_source_color and set_font_size?

Thanks!
Wayne

florianfesti, 2019-02-04:
There are some hidden reasons here. Guess the code would deserve some comments...

The white rectangle around text is not meant to be seen. It is a "fix" for another issue. I do calculated the size of the drawing afterwards. But the code doing this can't understand text. So the rectangle prevents the text being outside of the calculated area.

Overall the reference isn't meant to be cut or etched. It's purpose is to be able to check that the file has survived all the conversions without changing size. So I typically check it before cutting and then delete it in the laser software. That may not be a practical workflow for Pononko.

Overall I am wondering if named colors are really the right level of abstraction. May be we should have names for functions - like outside, holes, extra parts, etch cut, text, ... This could lead to having even more color schemes. E.g. while right now holes are colored blue having them black, too, would be even more convenient on out current laser cutter software.

wayn3w, 2019-02-04:
I didn't realize that the 'reference' was not to be cut.  Thank you for the explanation.  I can easily make do with passing --reference 0 to the script as part of my workflow.

Regarding colors, what would you think of a user-supplied palette?  By default each type of object would continue to get its own color, but a user can supply a file/dict that can specify the desired colors.  If there is no file, boxes can create one so the user can modify it and pass it to boxes.  

It's just a thought, and if it sounds reasonable, I can be of assistance writing the code.

florianfesti, 2019-02-06:
As the svg_Ponoko is intended to be used directly we could just force the reference param to 0 for that format.

Using a pallet file seems cumbersome. Yes, I just said we should implement the Ponoko variant as a format on its own... but we could also just offer a set of different palettes (that would include line width somehow). These would map names purposes to actual colors.

florianfesti, 2019-02-14:
8640514daf1bc68110d5f89f123e1541eb68cec9 disables the reference for the svg_Ponoko format. Let's leave it at this for now. We can do other color schemes later on.

closed 2019-02-14 by florianfesti