florianfesti/boxes #128: Feature idea: lids with handle

Blancanegra, 2019-03-09:
A handle type like the one showed in the image helps to place both lid pieces in perfect place.

Depending on the lenght and quantity of perpendicular spikes (to the optional handle) you use, would make glue unnecesary for the lid.

[image]

florianfesti, 2019-03-11:
I am not quite getting how this is supposed to look on the bottom side. Do those parts go all out to the outside walls? Do you have a picture showing the underside?

Blancanegra, 2019-03-12:
Sure, I just took some pictures showing different lids:

Top side:
[image]

Bottom side:
[image]

Pieces for the lid of the first post image:
[image]

Pieces for the lid of the hexagonal box:
[image]

florianfesti, 2023-04-07:
OK, I refactored the code for lids and added rectangular lids to all generators that really should have them. This includes infrastructure for handles. So far there is only one handle type supported and I don't have a good idea how to reuse the infrastructure for non rectangular boxes.
But adding more handle types and styles should be easy.

florianfesti, 2023-04-08:
Still need user documentation and (i) links.

Blancanegra, 2024-01-24:
Wow, those looks fantastic, thanks!

closed 2024-01-24 by Blancanegra