florianfesti/boxes #129: Box idea: sticky notes holder

Blancanegra, 2019-03-21:
I'm not a coder but would be very easy to create a generator for this kind of open boxes, cloning and modifying the stackable edge a bit for given sides of the box, including a radius setting:

[image]

Changing the size of the box it can hold sticky notes, business cards, napkins, etc . For napkins would be nice to have at least two opposite sides recessed.

Hope you like the idea!

ffesti, 2019-03-21:
We actually already have one of those cut for beer coasters. But it was done using a UniversalBox edited with Inkscape. Writing a generator should be pretty easy.

ZenithDK, 2019-03-21:
If you make the front a bit lower and add indents to the top sides, it
could even be stackable.

florianfesti, 2019-03-23:
c09ee50fbd8c6c8b8671acf6163c5fecc9e16da8 adds a simple version. I might add settings to the slots so you can select the width and radius later on. 
Not quite sure if stacking them is that of an interesting feature. But it should be easy to do by overloading the StackableEdge instead of a straight Edge.

florianfesti, 2019-03-23:
Any suggestions on the settings?

Blancanegra, 2019-03-24:
The most common square notes from different brands are 76x76 mm in size (3x3 inches).
My settings are 78x78 mm (for easy notes replacement) and 35 mm inner height for that notes size.

Blancanegra, 2019-03-27:
Testing notesholder.py and works fine, but  consider to add a slot holedistance parameter (together with width and radius) to make the wall stronger.

bwente, 2019-06-23:
Seems to me the finger holes are too close to the slot using notesholder.py, maybe omit the finger holes under the slot?

florianfesti, 2019-06-24:
Thinking a bit more leaving out the finger holes there is not really going to make a big difference. But having a cut out in the bottom will make it much easier to grab the last few sheets. But this requires cutting the front wall into two pieces.

bwente, 2019-06-24:
But the finger holes keep breaking.

[image]

Blancanegra, 2019-06-24:
> Thinking a bit more leaving out the finger holes there is not really going to make a big difference. But having a cut out in the bottom will make it much easier to grab the last few sheets. But this requires cutting the front wall into two pieces.

I prefer a strong wall than to easily keep the last few sheets. I'm modifying my notesholder with a slot to finger hole distance and making the slot wider up and narrow down:

[image]

florianfesti, 2019-10-19:
Ok, I have finally re-done the NotesHolder to be fully split in the front. Change is not on the we instance yet. Feedback is welcome.

closed 2019-10-19 by florianfesti

bwente, 2020-08-18:
Here is a desktop object that I created using the new Nots Holder combined with a "skinny" tray.

[image]

commit e2cb90db (2019-10-19): NotesHolder: Split front into two pieces and set better default values

Resolves: #129