florianfesti/boxes #131: Box Idea : tray storage

Shikoruma, 2019-03-30:
Wonderful works !

I love your tray generator. 
I think a tray storage would be nice. I think of that https://www.thingiverse.com/thing:2820385, with the option to add the bottom and the top. 
I think it is quite simple because it'is just a tray put verticaly with a big height and with or whitout the front and the back wall. 

I will try to build the generator.

florianfesti, 2019-03-30:
The easiest way is probably just extending the TypeTray a bit. It already can be rendered without the bottom/back wall by selecting the "e" Edge for the bottom.
Just add a outerwall param with ["all", "x_only", "y_only", "none"] (or may be even more fine grained or one axis only).

You have to replace the hard coded edges at the sides of the walls by variables containing "F" or "e" for the x walls and "f" or "e" for the y walls.

May be also add a description attribute to explain how these settings can be used, as the TypeTray is becoming much more than just a simple Tray.

Shikoruma, 2019-03-30:
Thank you for the advise.
I didn't subclasse TypeTray to keep the possibility to stack the boxes.
Here is my generator StorageTray https://github.com/Shikoruma/boxes/tree/storagetray
I will test it next week in my lab. If it works I will do a pull request.

florianfesti, 2019-05-11:
Any news? Is there anything I can help with?

Shikoruma, 2019-05-12:
Sorry, I finish it but I forgot to do the pull request. I make a picture tomorow and commit all of it. 
(During this work, I found that the outside option, for all boxes, is a little tricky depending on the edge you choose, maybe I will look into it later)

florianfesti, 2019-10-19:
This should be solved with PR #142 
Closing.

closed 2019-10-19 by florianfesti