florianfesti/boxes #134: Failed to run with docker.

MCuserme, 2019-04-08:
I build the project with docker by following instructions from the Dockerfile in scripts folder.
`docker build -t boxes.py .`
`docker run -ti -p 4000:8000 boxes.py`

And I get this error when trying to run an docker image.

`Traceback (most recent call last):
  File "boxes/scripts/boxesserver", line 41, in <module>
    import boxes.generators
ModuleNotFoundError: No module named 'boxes'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "boxes/scripts/boxesserver", line 44, in <module>
    import boxes.generators
  File "boxes/scripts/../boxes/__init__.py", line 23, in <module>
    import cairo
ModuleNotFoundError: No module named 'cairo'`

florianfesti, 2019-04-09:
Looks like there is a packaging issue in Fedora 29. python3-cairocffi actually needs python3-xcffib but is not required on a package level.
Added workaround in 83078d056f5ddcb321ec3bbeeabb4b536ef5f321

closed 2019-04-09 by florianfesti

florianfesti, 2019-04-09:
Filed at Fedora BZ as https://bugzilla.redhat.com/show_bug.cgi?id=1698217