florianfesti/boxes #135: Translation of PT-BR your Project!

vikingsbuild, 2019-04-11:
Hello, congratulations on your beautiful project!
I am not a programmer, but I understand something about phyton, here in BR they use their project a lot, but Brazilian has difficulty with the English language. I am translating into the PT-BR language and would like to know if you have the possibility to make future updates of your project, without losing the PT-BR translation.
I'm grateful!

florianfesti, 2019-04-11:
Hi!
Getting Boxes.py translated is a great idea. But I have very limited experience with i18n. Things are further complicated by the heavy use of meta programming in the Boxes.py UI. This means that strings are used for both UI but also program logic. The overall architecture of the web app may also cause issues.
So the first thing would be to choose a i18n library (probably gettext) and some translation tool/service. Do you have any experience with that?

florianfesti, 2019-07-24:
I started working on changes to make this happen (a while ago). I uploaded a  preliminary .pot file to https://translate.zanata.org/iteration/view/boxes.py/master
Not 100% sure if this is the way to go. Any advice on the proper tool to use is welcome.

florianfesti, 2019-08-22:
Ok, the code for translating the web interface is live. For pt_br one would need to translate the .pot file on zanata (see link above) and drop me a note here.

The current implementation uses browser preferences. At some point I might add some preferences. Otoh I kinda like the way the web site is currently completely stateless and doe snot require cookies.
The only language currently supported is zh_CN (Chinese).

florianfesti, 2019-08-24:
Of course it is also possible to just download the POT file and translate it locally: https://github.com/florianfesti/boxes/blob/master/po/boxes.py.pot
Just append the result here or open a separate ticket.

florianfesti, 2019-09-22:
I am closing this ticket for now. The infrastructure in the code is all there and the interface can be translated to any language. Feel free to re-open as soon as there is some work done on the pt-BR translation.

closed 2019-09-22 by florianfesti