florianfesti/boxes #139: Empty svg output

DomingoMolinero, 2019-05-01:
Hi,

first, thank you for this powerful tool, I love it!

but now the bad news: After having created maybe a hundred boxes, I am randomly unable to render boxes. I experienced the problem at first on april 18th. For example using https://www.festi.info/boxes.py/ClosedBox with the predefined values leads randomly to an empty svg file. I haven't tried all scripts, but the 10-15 ones i did were similar in failing in about 75% of the cases. 

Altering values does not change anything in the behaviour and failure rate. Problem occures on 3 different PCs.

Any idea?

MCuserme, 2019-05-07:
Also empty .dxf files. (tested on ClosedBox)

florianfesti, 2019-05-11:
Duplicate of #136

closed 2019-05-11 by florianfesti