florianfesti/boxes #140: Pictures missing

THX2112, 2019-05-10:
Many of the existing pictures are missing, and a lot of boxes don't have pictures or renderings. This makes it hard to figure out what the parameters do, or what the final box should look like without a lot of trial and error.

florianfesti, 2019-05-11:
Yeah, making good pictures is difficult and a lot of work. I am especially bad at taking photos. So if you could take some and either open an PR or attach them here that would be a great help.

Blancanegra, 2019-11-02:
I took some photos, please use it at your own convenience.

ElectronicsBox, and opened and closed:
[image]
[image]

PaintStorage first version, stacked and with paint cans:
[image]

LaserHoldFast in action (ugly set but best than nothing):
[image]

NotesHolder (default values):
[image]

florianfesti, 2019-11-02:
Thank you very much! Added as 9a8ae19 together with some picture of the WineRack I still had somewhere.

florianfesti, 2019-11-02:
Ok, this ticket is kinda useless unless we actually track the missing pictures. So here's a list:

- BayonetBox
- Rack*Box
- FlexBox5
- HeartBox
- StorageRack
- Slatwall* - need a photo session in our wood workshop
- BurnTest
- ConcaveKnob
- FlexTest - could be rendered image
- FlexTest2 - could be rendered image
- GearBox
- Gears
- LBeam
- Planetary
- Planetary2
- Pulley
- RectangularWall
- WaivyKnob 
- AngledCutJig
- Arcade
- Display
- DrillBox
- Hook
- LaserClamp
- MagazinFile
- MakitaPowerSupply
- OttoLegs - get from Ottobot project page
- OttoSoles - get from Ottobot project page
- RoyalGame
- Rotary

Wolfpuppie, 2020-07-10:
I took a picture of a 'console' box:

[image]

here are some other angles: https://imgur.com/a/soQyL51

florianfesti, 2020-07-14:
Thanks! See https://festi.info/boxes.py/Console?language=en for the new image!

DiegoLascano, 2020-08-12:
has anyone built the rotary? it is kinda difficult to understand since there are lots of parts.

florianfesti, 2020-08-12:
> has anyone built the rotary? it is kinda difficult to understand since there are lots of parts.

Yes. But as it currently is it's not that great and should probably be improved before anyone else builds one.
The contraption consists of 3 parts that are put on an aluminium profile:

At the very top of the drawing are the parts for the motor block. It has two rollers that are driven by an NEMA 23 stepper by a timing belt. The pulley for the motor itself is not included as the one I used already had one.

Below is a second block/box that is intended to hold the aluminium profile at the other end. Both of them have hooks to allow hanging the whole thing between to knifes of an knife table. This in theory saves a bit of usable height but in practice is just too much of an hassle and the hooks are too fragile anyway. So this pieces is not even needed.

The bottom half of the drawing (starting with the row of small rectangular pieces) is a scissor lift carrying to freewheeling rollers. The rollers are shown in the drawings but must be deleted before cutting. The problem is that the scissor lift is just too complicated and not rigid enough. One solution would be to just have a box with two rollers that slides on the aluminium profile and angle the whole thing instead of raising and lowering only the rollers.

The whole thing also misses a couple of features regularly found on professional rotary attachments like adjustable stops and rollers holding down the work piece.

If you want to build one I might just program some improvements. This is more likely to happen if you have an idea what features you want. Often making decisions on how things should look like are much harder than actually make them look that way. Alternatively you can just build the motor block twice and leave the rollers on the second block turning freely.

florianfesti, 2020-08-17:
[image]
Here's a picture. I'll add it to the generator, too. It still doesn't show very clearly how the scissor lift is put together. But you really shouldn't anyway...

Wolfpuppie, 2020-08-27:
Heartbox: https://i.imgur.com/PqfBcZX.jpg

florianfesti, 2020-08-30:
> Heartbox: https://i.imgur.com/PqfBcZX.jpg

0403f01f0ed1c26d53848e739c641ba0b0f6eb88

Thanks for the picture!

oscaracena, 2020-09-29:
These two may also be useful, although no the best quality...

[image]
[image]

florianfesti, 2020-09-30:
> These two may also be useful, although no the best quality...> 
Well, still a lot better than no image at all. Added as 928e88f070ffcf14d3afc364408dfe92d99e5cb2 Will update the web instance soon.
Thanks a lot!

Blancanegra, 2020-10-07:
Some shots and comments:

BayonetBox
[image]

Placing perimetral holes to insert metal wire or wood sticks would be helpful for aligning the segments when gluing them.

[image]

BasedBox
[image]

Would be possible to add an option to customize the base size (hole distance?)?

LBeam
[image]

In some cases like that, is not easy to figure what x, y and h parameters really are, so I engraved those in the corresponding sides for the picture.

impellerturn, 2020-12-29:
Hi,
here some pictures of StorageRack:
[image]
[image]
[image]
Best regards

florianfesti, 2020-12-29:
Thank you very much! Added with e6dccc413555817335786a815b42b57153039b83

Arcturuss, 2021-03-21:
> In some cases like that, is not easy to figure what x, y and h parameters really are, so I engraved those in the corresponding sides for the picture.

> Yes, I should rename all the dimensions at some point to meaningful names.

Maybe it'll be more useful to show all dimensions on photo or drawing, especially in generators that are not so obvious. 
smth like this:

[image]

ClaneSharke, 2021-06-18:
Images for Angle Cut Jig
[image]
[image]
[image]
[image]
[image]

tzwilliams, 2021-09-24:
Love the tool!! Here are two **RoundedBox** images showing different `edge` styles.

Default: `edge_style = f Finger Joint`
[image]

Alt: `edge_style = h Edge (parallel Finger Joint Holes)`
[image]

tzwilliams, 2021-09-24:
Some additional ShutterBox images (using the spring finger joint option and with sanded corners)
[image]
[image]
[image]

florianfesti, 2021-09-26:
Added with d6c8d965c3a4bca7c88739774161f1ab281c44ce and cbbe8c11cc781fade7286ec323f5a93798269075

@tzwilliams : I kept the original image for the ShutterBox as it shows an un-sanded example. That's a bit of a shame as your picture is much better. But the old one fits better into the sample pictures which are all un-sanded.

Anyway, thanks for these very nice pictures!

tzwilliams, 2021-09-26:
@florianfesti, my pleasure.  I agree with keeping the old pictures--I think they each show useful information.  I'll try to keep contributing pics as time allows :)

profparedes, 2021-10-13:
TriangleLamp

Day0Dreamer, 2022-04-12:
TwoPiece
[image]
[image]
[image]

vinicius-bianco, 2022-04-29:
Hi, i don't spend a lot of time doing this, so the render isn't good. I will try to make a better render, but maybe this one is useful... here is my default Drill Box, rendered in eevee, blender.
[image]

florianfesti, 2022-04-30:
> Hi, i don't spend a lot of time doing this, so the render isn't good. I will try to make a better render, but maybe this one is useful... here is my default Drill Box, rendered in eevee, blender. [image]

This is "assembled" upside down. The larger ridge should be at the top, to give a bit of protection to the drills. Otoh there is not much use to have this amount of space below the bottom.

florianfesti, 2022-04-30:
> Images for [drillstand ](https://www.festi.info/boxes.py/DrillStand?language=en) with some steps for assembly. Careful with the orientation of the bottom part (B), it's slightly asymmetrical.

Added your pictures to the description. Thanks a lot! This makes assembly a lot easier to understand.
Turned out I actually had added a sample image a while a go but with the wrong name m)

florianfesti, 2022-04-30:
> TwoPiece, default settings 10cm x 10cm x 10cm outer borders; 2.5mm thick

Added one of the pictures to the description - together with a bit help on the `hi` param. As the box is very simple it does not really justify more images. Still thanks!

vinicius-bianco, 2022-05-03:
> 
:)
[image]

vinicius-bianco, 2022-05-03:
> This is "assembled" upside down. The larger ridge should be at the top, to give a bit of protection to the drills. Otoh there is not much use to have this amount of space below the bottom.

fixed and improved :)

[image]

HaSHsss, 2022-06-06:
Made  in Photoshop  for `AllEdges.py` generator

[image]

florianfesti, 2022-06-07:
Thanks! Added with 4344c93b6ef. Will update the web instance as soon as I am back home. Don't want to touch it while being at the airport.

florianfesti, 2022-06-08:
Added with aba88f3e1da9e572cf5578536cfa5b3b39439f2b. Thanks a lot!

ednisley, 2022-12-01:
A small BreadBox:

[image]

The door works great:

[image]

A few more details on [my blog](https://softsolder.com/) at https://softsolder.com/2022/12/01/rolltop-cupcake-box/

florianfesti, 2023-01-13:
Oh, this post got lost in pre x-mas rush. Finally added with d3df27b238db27310b42e801b25926184fd4985a
Cutting and taking pictures of new generators is super helpful. Thanks @ednisley !

florianfesti, 2023-02-18:
@eoto88 Added as 71e23910ccf47b566bd62326566ce034fd56be80 

Thanks for the picture!

d5e, 2023-02-26:
**Drillbox**.

Made of 3mm MDF with a 150W CO2-Laser. Glue-Free :)

[image]

florianfesti, 2023-02-26:
Added with 0e4e373367059cd165562cb6f1a8e0c86f43ee32. Thanks for the picture!

datazaShane, 2023-03-30:
ConcaveKnob
[image]
[image]
[image]

datazaShane, 2023-03-30:
Laptop Stand
[image]
[image]

datazaShane, 2023-03-30:
Used Drillbox Template to make box for Router Bits
[image]

datazaShane, 2023-04-23:
Hook
[image]
[image]

florianfesti, 2023-04-23:
Hmpf, guess Ad with one "d" work here, too...

PhoenyxAshe, 2023-05-02:
I just cut and partly assembled a small version of the triangle lamp, and once I've had a decent sleep and figured out the probably quite obvious steps I'm missing, I'll post the updated version.  I'm guessing either dowels in the screen holes and a flexible material wrapped around, or screen panels between the two ends?  Both?
[image]

florianfesti, 2023-05-02:
> I'm guessing either dowels in the screen holes and a flexible material wrapped around, or screen panels between the two ends? Both?

Yeah, this generator is clearly missing some documentation. It was done as a project for another institution and ofc they knew what they wanted to do...

florianfesti, 2023-05-02:
Ehm, I mean: The idea is to use translucent acrylic for the top triangle and screw that in place. The screws should pass through the top layer but bite into the little wooden triangles underneath. IIRC they also added connectors to the sides so that multiple triangle lamps could be connected to a larger installation.

BeppieTheKlapp, 2023-11-18:
Got two
[image]
[image]
for the 7-segment digit display:

crashnbernstein, 2024-02-13:
Pin Rack render
[image]

florianfesti, 2024-02-13:
Added with 10012e596694ecd9ce Thanks!

florianfesti, 2024-02-13:
This is getting too long and unwieldy. Created a new issue for this: #628

closed 2024-02-13 by florianfesti