florianfesti/boxes #141: Sliding tray lid error

DrakeCoker, 2019-05-10:
When creating a tray with unequal-size compartments and a sliding lid, the resulting box has the two locking slots on different sides of the box.  You can't tell with equal-size compartments, but with unequal, it quickly becomes apparent that the box cannot by put together!

Example: https://www.festi.info/boxes.py/TypeTray?FingerJoint_angle=90.0&FingerJoint_surroundingspaces=0.5&FingerJoint_edge_width=1.0&FingerJoint_finger=2.0&FingerJoint_play=0.0&FingerJoint_space=2.0&FingerJoint_width=1.0&Stackable_angle=60&Stackable_height=2.0&Stackable_holedistance=1.0&Stackable_width=4.0&Hinge_grip_percentage=0&Hinge_outset=0&Hinge_pinwidth=0.5&Hinge_style=outset&Hinge_axle=2&Hinge_grip_length=0&Hinge_hingestrength=1&CabinetHinge_bore=3.2&CabinetHinge_eyes_per_hinge=5&CabinetHinge_hinges=2&CabinetHinge_style=inside&CabinetHinge_eye=1.5&CabinetHinge_play=0.05&CabinetHinge_spacing=2.0&Lid_angle=90.0&Lid_second_pin=0&Lid_second_pin=1&Lid_spring=none&Lid_surroundingspaces=2.0&Lid_edge_width=1.0&Lid_finger=3.0&Lid_play=0.05&Lid_space=2.0&Lid_width=1.0&Click_angle=5&Click_bottom_radius=0.1&Click_depth=3.0&RoundedTriangleEdge_height=150.0&RoundedTriangleEdge_r_hole=2.0&RoundedTriangleEdge_radius=30.0&RoundedTriangleEdge_outset=1&sx=10%3A10%3A10%3A30&sy=50*2&h=100.0&hi=0.0&outside=0&outside=1&bottom_edge=h&top_edge=L&gripheight=0&gripwidth=0&thickness=3.0&format=svg&tabs=0.0&debug=0&reference=100&burn=0.1&render=1

florianfesti, 2019-05-10:
The slots are obviously correct. It's the other holes on the side wall that should be mirrored.
Need to add a param to the callback...

DrakeCoker, 2019-05-10:
Well, duh, you are obviously correct!  Thanks for taking a look at the issue.

closed 2019-05-11 by florianfesti

florianfesti, 2019-05-12:
Still need to update the server instance, though...

DrakeCoker, 2019-05-28:
Curious when the next server update might happen?  I noticed today that the flaw is still in the production version.  Thanks!

commit a6ca7237 (2019-05-11): Mirror holes on the other walls

to have them match up. Otherwise boxes with non symetric top edges don't fit
together.

Fixes #141