florianfesti/boxes #143: 0 byte files from online generator

gregpinero, 2019-05-23:
I'm trying this and the output seems to be empty:

https://www.festi.info/boxes.py/IntegratedHingeBox?FingerJoint_angle=90.0&FingerJoint_surroundingspaces=2.0&FingerJoint_edge_width=1.0&FingerJoint_finger=2.0&FingerJoint_play=0.0&FingerJoint_space=2.0&FingerJoint_width=1.0&ChestHinge_hinge_strength=1.0&ChestHinge_pin_height=2.0&ChestHinge_play=0.1&x=171.5&y=134.75&h=144&outside=0&outside=1&lidheight=34&thickness=3.0&format=svg&tabs=0.0&debug=0&reference=100&burn=0.1&render=1

matthijskooijman, 2019-06-17:
I've seen this happen a few times today as well. A reload usually fixes things, though I've also seen invalid XML after a reload and then a valid SVG after a second reload. I haven't saved the error though, and now can no longer reproduce the error (I can reproduce the empty SVGs).

florianfesti, 2019-08-16:
Ok, I fixed some tracebacks I found in the server logs. It is possible that they made the worker defunc until they got restarted. I am closing this now in the hope things go smoother from here on. Please re-open if you run into this issue again.

closed 2019-08-16 by florianfesti