florianfesti/boxes #147: Box idea: asymmetric flexbox

DedeHai, 2019-06-22:
How complicated is it to modify the 'flexbox' to make a box with two different radii so there will be a slope so it looks like the envelope of two circles of different size. I mean somthing like this:
[image](http://www.yenka.com/activities/Calculating_Speed_Reduction_and_Increase/attachments/Calculating%20Speed%20Reduction%20and%20Increase%20-%20model1.gif)

Would this be an easy tweak or require a complete rewrite?

florianfesti, 2019-06-24:
I would probably copy the generator to create a new one as FlexBox is basically rectangular with rounded corners (sides defined by x,y, r). The box you suggest will be completely rounded so the sideds will be defined by x, r1, r2. The changes to the actual code is not too big. Some angles and length need to be calculated. But that's nothing that middle school maths can't do.

florianfesti, 2019-07-11:
Generator is still untested and missing a picture. So if you can test it and make a good photo that would be of great help.

commit 88e99a57 (2019-07-11): New generator: FlexBox5

Resolves #147
closed 2019-07-11 by florianfesti