florianfesti/boxes #154: Add explanation of how to create own boxes to documentation

PellelNitram, 2019-07-25:
Hi,

I recently found this repo and liked the idea right away! Excellent work so far!

I was looking for an explanation of how to create own boxes in the documentation. I believe it would be very useful as a quickstart guide on how to add custom generators.

Best wishes

florianfesti, 2019-07-25:
https://florianfesti.github.io/boxes/html/apidoc.html has the docs needed. Have a look at the "Examples" sub page. It gives pointers to easy to understand box generators. That's probably the easiest way to get a grasp on how to get started.

If you have specific questions feel free to ask them here or open a ticket asking for a specific topic / section you want to be added to the docs.

matthijskooijman, 2019-07-26:
I believe that those examples show how to add generators by modifying the boxes.py source tree (e.g. adding files there). I have been looking for a way to define boxes / generators *outside* of the boxes.py tree (e.g. so you have a  single .py file that you can distribute, which works as long as boxes.py is installed in the pythonpath). Maybe this is also what @PellelNitram is looking for?

I ended up copying some code from the boxes.py main function, which works but feels a bit ugly (since it duplicates code). This might be easy to fix by adding an API function that takes a generator (`Boxes` subclass), and renders it (possibly also parsing cli arguments)?

florianfesti, 2021-05-23:
All you have to do is dropping the file into `boxes/generators`. I am not sure I want to add a feature to look elsewhere. Closing.

closed 2021-05-23 by florianfesti