florianfesti/boxes #157: Export to DXF not precise

kolins-cz, 2019-07-28:
Hi,
Trying web version and when I import .dxf to my software, 100mm reference square is actually 100.0002mm. It won't be an issue for most, but I would like to convert my box to 3D and make some changes before manufacturing. And this imprecision complicates 3D design. Something must be wrong with DXF exporter.

Way to get around this is to export .svg, open in Inkscape, export to .dxf and open. After that, line is really 100.0000mm

I like this software and I hope this can be fixed easily.
Thanks

florianfesti, 2019-07-28:
The DXF is currently produced by generating Post Script with the PS backend of the cairo library (https://www.cairographics.org/backends/). It is then converted to DXF using ps2edit. One of these steps obviously messes things up. Looks like somewhere single precision floats are used. But this all happens outside of the Boxes.py code. So Boxes.py can not fix this easily

To be honest getting DXF export working was not as easy as I had wished for. So I am currently pretty happy it works at all. If you can suggest any other tools for converting postscript/PDF or (preferably) SVG to DXF I am happy to try them out. Running Inkscape on the web server otoh is not an option.
But without switching to other tools your only option is to convert the SVG yourself using Inkscape.

florianfesti, 2019-08-16:
I am closing this as WONTFIX. If there is a easy way to switch out the tooling I am happy to do so but I currently don't really see a way to get this fixed.

FYI there is some work lurking around on replacing cairo by self written back end. This may allow writing a proper DXF back end that is not relying on on pstoedit and though solve this issue. But this is all too much in the future to keep this issue open waiting.

closed 2019-08-16 by florianfesti